tbnobody/OpenDTU #256: Feature Request: Blindleistung regeln, wie es die orig. DTU kann

bluespiano, 2022-10-22:
Vorab allen Beteiligten mein herzliches Dankeschön für den Projektfortschritt.
Ich kann zwar nicht programmieren, aber vielleicht durch meinen Hinweis das Produkt noch besser machen .. ?

In den Beschreibungen zu Wechselrichtern der HM-Serie und auch DTU steht: "Leistung- und Blindleistung regelbar (DTU benötigt)"
Über zB Ahoy wird bei einigen Usern ein Blindleistungswert von >5 bis zu 20 oder 30VAr angezeigt.
Die Blindleistung sollte immer mögl. niedrig sein, damit sie quasi zu Wirkleistung wird.
Die originale DTU bietet anscheinend genau hier eine manuelle Regelung, die ich bei Ahoy und auch openDTU vermisse.
Gibt es da schon Erkenntnisse?

homeautomation2022, 2022-10-23:
Es scheint von der Gridfile auf dem Gerät abhängig zu sein, also je nachdem für welches Land das Gerät ursprünglich mal vorgesehen war. Die GPF kann man in der Hoymiles Cloud auswählen und laden oder auch selbst anpassen(unzulässiger Betrieb an unserem Stromnetz möglich!).
Meiner ist jedenfalls auf cos phi = 1,0 festgenagelt, er lehnt das CMD zum setzen einer Blindleistung oder eines anderen PowerFactors ab. Für mich nicht schlimm, da ich ja eh nur 100% Wirkleistung möchte...
Die VDE-AR-N 4105 hat dazu auch Vorgaben, die letztendlich in der GridProtectionFile eingepflegt sind. Ob der PF wirklich in der GPF steht kann ich nicht genau sagen, aber die erlaubten Spannungs- und Frequenzbereichebereiche auf alle Fälle, da wäre die Blindleistung naheliegend.

sbiede, 2023-07-30:
> Es scheint von der Gridfile auf dem Gerät abhängig zu sein, also je nachdem für welches Land das Gerät ursprünglich mal vorgesehen war. Die GPF kann man in der Hoymiles Cloud auswählen und laden [...]

>>"Meiner ist jedenfalls auf cos phi = 1,0 festgenagelt, er lehnt das CMD zum setzen einer Blindleistung oder eines anderen PowerFactors ab. "
@homeautomation2022  welches CMD hast du denn abgestezt um die Einstellung zu verändern?

fwelzel, 2023-09-28:
Gibt es hierzu schon eine Lösung, da mein Strom Netzbetreiber  einen COS phi von 0,95 fordert?

k1llerk3ks, 2023-09-28:
@fwelzel Uns wurde ebenso mitgeteilt, dass die Anlage einen Cos phi von 0,95 benötigt. Wir haben 2kW, weswegen laut der Recherche dennoch ein Cos Phi von 1 in Ordnung ist, laut dieser Quelle:

> Abhängig von der Größe der Photovoltaikanlage fordert die VDE-Anwendungsregel Werte von 0,95 untererregt bzw. übererregt bis 0,90 untererregt bzw. übererregt für cos(ϕ): Anlagen bis 3,68 kVA müssen keine Blindleistung einspeisen, arbeiten also mit einem cos(ϕ) von 1. Anlagen zwischen 3,68 kVA und 13,8 kVA müssen Blindleistung mit einem cos(ϕ) von 0,95 und alle Anlagen über 13,8 kVA mit einem cos(ϕ) von 0,90 einspeisen.

Siehe https://www.geoplex.de/fileadmin/user_upload/PDF_PV/Gesetzgebung_Photovoltaik_seit_2012.pdf

Bei einem Cos Phi von 1 ist 1 kvA = 1kW.

Kann das jemand bestätigen?

rocket-plane, 2023-10-09:
Ich würde mich ebenfalls über diese Funktion freuen, da ich für die Anlage (>3.6kVA) den cos(ϕ) einstellen muss und ansosten wohl nur dafür eine original DTU kaufen und einrichten müsste.

stefan123t, 2023-10-09:
Ja man kann das bestätigen.

Hier haben wir die Kommandos zum Regeln des WR dokumentiert. Da gibt es ggf auch ein ReactivePowerContr (0x0C) Reactive Power Limit

https://github.com/lumapu/ahoy/wiki/Protocol#welche-device-control-devcontrol_all-0x51-sub-kommandos-gibt-es

Weiter unten sind auch die von @homeautomation2022 angesprochenen Grid Profiles u.a VDE4105_2018 dokumentiert.

https://github.com/lumapu/ahoy/wiki/Protocol#file-name-de_vde4105_2018

Dort steht mE bei allen Power Factor basierten Optionen 0 / disabled im Profile. Müsste man also vorher das GridProfile Update Command tracen und implementieren. 

Cave Canem / Vorsicht vor dem Hund: 
Dafür sollten wir erstmal das Grid Profile auslesen können, bevor wir es drüber flashen können und wollen.
Siehe auch Issues #1369, #987, #707, #900, #679 sowie die Discussions #1394, #364, #550.

-- | -- | -- | -- | --
46 | Specified Power Factor (SPF) (?) |   |   |  
47 | SPF Function Activated | 0 |   | 0~1
48 | Power Factor (PF) | v0.95 |   | 0.9~1
49 | Watt Power Factor (WPF) (?) |   |   |  
50 | WPF Function Activated | 0 |   | 0~1
51 | Start of Power of WPF (Pstart) | 50 | %Pn | ~
52 | Power Factor ar Rated Power (RFRP) | 0.95 |   | 0.8~1
53 | Reactive Power Control (RPC) (?) |   |   |  
54 | RPC Function Activated | 0 |   | 0~1
55 | Reactive Power (VAR) | v 0 | %Sn | 0~50

stefan123t, 2023-10-09:
In Ahoy wurde das ganze bisher in https://github.com/lumapu/ahoy/issues/96 und vor allem https://github.com/lumapu/ahoy/issues/365 analysiert.

JsBergbau, 2023-12-13:
>  Die Blindleistung sollte immer mögl. niedrig sein, damit sie quasi zu Wirkleistung wird.

So ganz kann ich das nicht stehen lassen. Bei Wechselstromnetzen kann es sein, dass man Blindleistung einspeisen muss, damit die Spannung nicht unnötig hoch wird. Das war damals auch ein Grund mich für den HM-1500 zu entscheiden, damit ich da die Blindleistung einstellen kann, sollte die Netzspannung zu hoch sein und er dann nicht abschalten muss. Sehr ausführlich ist das in diesem Video erklärt https://www.youtube.com/watch?v=BwP1QzIq4eE
Das dürfte auch erklären, warum die Blindleistungseinspeisung bei größeren Anlagen gefordert ist.

stefan123t, 2024-07-10:
@tbnobody der Feature Request hier ist in #900 speziell [hier detailliert](https://github.com/tbnobody/OpenDTU/issues/900#issuecomment-2209546838) beschrieben.

Dort ist es ein Feature Request um ein bekanntes Grid Profile an den WR zu senden.
Der zweite Schritt wäre dann die in den Grid Profiles Werte in den angebenen Ranges frei wählbar zu machen um das Grid Profile individuell anzupassen.

JsBergbau, 2025-04-03:
Immer noch aktuell...

JsBergbau, 2025-05-05:
Weiterhin relevant.

stefan123t, 2025-05-21:
@tbnobody bitte mit enhancement taggen. Danke

stefan123t, 2025-06-21:
@JsBergbau ich kann die Frage verstehen, da es technisch möglich sein müsste dies zu regeln, sollte OpenDTU das auch unterstützen. Fir Anpassung des Commands sollte nicht so schwierig sein und auch für das GridProfile haben wir ggf. bereits ein Tracefile. Hier gibt es die größten Bedenken dass man evtl ein für das Stromnetz ungeeignetes / unpassendes GridProfile einspielt. Zb. Island Protection 0 bzw. NA-Schutz deaktiviert!

Was für einen praktischen Nutzen / Anwendungsfall hast Du bei dem Du > 1kVA "einspeisen" willst mit einem Microwechselrichter der für Balkonkraftwerke bis 800 W gedacht / gedrosselt ist ?

rocket-plane, 2025-06-21:
@stefan123t: Hier mal mein Anwendungsfall:
Unsere ausgewachsene PV Anlage (ca 15kWp) besteht aus Verschattungsgründen (und weil die OpenDTU so gut funktioniert) aus vielen HMS2000-4T Wechselrichtern. Der Netzbetreiber fordert, dass diese mit einem Cos(phi) von 0,9 einspeisen sollen. Zusätzlich hängen die Wechselrichter am AC Out von 3 Victron Multiplus. Der Victron steuert die Leistung im Inselbetrieb dabei über die Netzfrequenz des Inselnetzes. Dazu muss die Frequency Watt Funktion der Hoymiles richtig eingestellt werden.

stefan123t, 2025-06-23:
> @tbnobody der Feature Request hier ist in #900 speziell [hier detailliert](https://github.com/tbnobody/OpenDTU/issues/900#issuecomment-2209546838) beschrieben.
> 
> 
> 
> Dort ist es ein Feature Request um ein bekanntes Grid Profile an den WR zu senden.

@rocket-plane @JsBergbau schaut Euch bitte den oa. Link an.

https://github.com/tbnobody/OpenDTU/issues/900#issuecomment-2209546838

Dort ist es ein Feature Request um ein bekanntes Grid Profile an den WR zu senden.Das sind 8 Commands 0x0A mit 0x01..0x07/0x88 und das letzte wird 20 Sekunden lang wiederholt.

> Der zweite Schritt wäre dann, die in den Grid Profiles gesetzten Werte in den angebenen Ranges frei wählbar zu machen, um das Grid Profile individuell anzupassen.

Der zweite Schritt wäre etwas kniffliger da wir hierzu das Grid Profile in bestimmten Wertebereichen anpassen müssen.

Hierzu bräuchten wir einen Editor der einmal die editierbaren Felder aus dem GridProfile Parser kennt und Eingabefelder zur Verfügung stellt und andererseits die Wertebereiche dieser Felder kennt und ggf. prüft.

Wenn wir hingegen ein gültiges Grid Profile haben können wir das einfach auswählen ggf. die CRC16 als Checksumme berechnen und dann wie oben das bekannte Grid Profile in acht Paketen (mehr oder weniger je nach Grid Profile) an den WR senden und die Checksumme ca. 20 Sekunden lang wiederholen.

Der selbe Code könnte prinziell auch für das Feature zum Firmware Update des Wechselrichters genutzt werden. Hier haben wir aber bisher noch nicht ausreichend Firmware Abbilder (nur ein/zwei Versionen für ein bestimmtes Modell) gesammelt.

clatoverada, 2025-08-16:
Thema beim Akkudoktor:
https://www.youtube.com/watch?v=O5xdGkpe9xE

AlexJacu, 2025-08-17:
Es wird immer wichtiger diese Parameter anzupassen, bei uns steigt die Spannung immer öfter auf 250v. Der Wechselrichter hat deswegen schon öfter abgeschaltet. mit der Blindleistung könnte das verhindert werden.

stefan123t, 2025-09-25:
@github-actions das ist ein Feature Request, der kann m.E. nicht stale werden. Die Grundlagen zum Grid Profile Update sind im #900 beschrieben. Irgendwie kann man das ReactivePowerLimit auch setzen, zumindest geht das aus dem Source Code bei gitee.com und den alten Dokumentationen zur Hoymiles S-Miles Cloud hervor.

AlexJacu, 2025-11-30:
If it will be implemented, please allow it to set like AC Limit. temporary and fixed.

AlexJacu, 2026-04-01:
Reopen case, still not solved