tbnobody/OpenDTU #266: ESP32 won´t connect to the internet

mowegener, 2022-10-25:
I can connect via the build-in AP to the ESP32 and can configure in OpenDTU, but the AP always closes itself after a certain time, ofcourse after the three minutes, but it opens and closes randomly. I configured the new internet connection but when I search up the ip address, it says ERR_CONNECTION_REFUSED. Has anyone an idea why it does this, I would be fine if I could just connect to it via the old ip(http://192.168.4.1), but I cant delete the network settings. Maybe Ive forgotten something... 
I hope someone got an idea.

mowegener, 2022-10-25:
Meanwhile I noticed another problem, right now I can log into the AP, ive set up the inverter(2022), but it isnt recognized, I was about 5 metres away from the inverter and restarted the inverter etc.. But it wont work, I think that it has to to with the wifi, do I really HAVE to buy the NRF24L01+ board or could it work with the antenna I use right now? Or do I just have to change some settings so itll work? I dont know what to do right now. My question is, if I buy the radio board, would it even work with my ESP32, because like mentioned earlier, I dont have the exact same board. I even think that the pins arent located in the same spots or if there are even the same pins.

madmartin, 2022-10-26:
Hello @mowegener ,

did you look at the serial log of the ESP32?
When you see the AP re-appear after a certain time, I guess the ESP crashes and restarts. There is nothing in the code that re-opens the AP at usual runtime.
Then, next topic, if I understand you right you have ONLY an ESP32 board? Did you read the README.md of the project? Of course you need an NRF24L01+ board, this board implements the radio technology to talk with the hoymiles inverter. This has nothing to do with any questions of internal or external antennas on any of the boards. You can think of the NRF24L01+ speaking a different radio language compared to the WiFi part of the ESP32.

mowegener, 2022-10-28:
Hello @madmartin ,
thanks for replying to my issue, ive already bought the NRF24L01+ board, right now i have the next issue, the esp32 wont connect to my internet. Just to test, ive tried to connect in my phone(hotspot), it worked perfectly fine! But I really want to connect it to my main internet, so I can see it from all the pc´s in my network. When im connecting to my router via wifi, the first 5 seconds it always says that there is no internet. Maybe thats an issue? I resetted it etc., it just wont work, the log is completely empty. My Esp32 even says that it found the board. I dont know what to do anymore,

mowegener, 2022-11-05:
It isnt finding the inverter aswell!

madmartin, 2022-11-06:
> When im connecting to my router via wifi, the first 5 seconds it always says that there is no internet. Maybe thats an issue? 

Hmm, in which way openDTU says "no internet"? I've checked the source code, the word "internet" does not appear in the source. Please explain

>  the log is completely empty.

the log of an ESP is never empty. After pressing the reset button, at least the bootloader spits out several lines. So thats a configuration issue with your terminal program. 

> ive tried to connect in my phone(hotspot), it worked perfectly fine! 

Thats good. Then you have to fill in 'Settings->Network' your WiFi parameters and save. After save, go to "Info->Network' (still using your phone)  and check the information on "WiFi Information (Station)" for Quality/RSSI and on "Network Interface (Station)" if the IP adress data makes sense.

mowegener, 2022-11-11:
Neuer Ansatz, ich habe mittlerweile festgestellt, dass das Problem am Router bzw. am Internet liegt. Sowohl mit Handy als auch mit dem HotSpot am PC, welcher mein WLAN einfach nur weiterreicht, funktioniert die Verbindung einwandfrei. OpenDTU verbindet sich direkt ohne Probleme. Andes als bei meinem WLAN, evtl. liegt es am Kanal. Aktueller Kanal im 2,4 GHz Bereich ist der Kanal 11. Habe auch schon von Geräten gehört, welche nur mit Kanälen über 100 funktionieren. Eine andere Idee habe ich mittlerweile auch nicht mehr.

tbnobody, 2022-11-11:
Spontan habe ich dazu nur das hier gefunden: https://olimex.wordpress.com/2021/12/10/avoid-wifi-channel-12-13-14-when-working-with-esp-devices/

Aber dort wird nur von Channel 12-14 geschrieben.

madmartin, 2022-11-11:
> Habe auch schon von Geräten gehört, welche nur mit Kanälen über 100 funktionieren. 
Das ist WLAN im 5 GHz  Bereich. ESP32 kann nur 2,4 GHz. Daran kann es also auch nicht liegen - außer dein WLAN Router macht ausschließlich 5GHz.
Bei 2,4 GhZ Kanal 11 zu nehmen ist unproblematisch. Die Warnung vor den Kanälen 12-14 kommt daher daß diese Kanäle nicht in allen Ländern dieser Welt für WLAN freigegeben sind. So ist man mit Kanal 1-11 bei allen Geräten im Netz immer auf der sicheren Seite.

Was ist denn bitte mit dem seriellen Konsole-Log vom ESP32? Was passiert da?

mowegener, 2022-11-12:
Meinen Sie diesen Log? Habe mich noch weiter belesen, aber ein weiterer Log ist mir nicht bekannt.

18:07:14.984 > WiFi disconnected
18:07:14.984 > Try reconnecting
18:07:14.984 > Network lost connection
18:07:14.984 > WiFi disconnected
18:07:14.984 > Try reconnecting
18:07:14.984 > Network lost connection
18:07:16.241 > WiFi disconnected
18:07:16.242 > Try reconnecting
18:07:16.242 > Network lost connection
18:07:19.091 > WiFi disconnected
18:07:19.091 > Try reconnecting
18:07:19.091 > Network lost connection
18:07:21.934 > WiFi disconnected
18:07:21.939 > Try reconnecting
18:07:21.939 > Network lost connection
18:07:24.785 > WiFi disconnected
18:07:24.786 > Try reconnecting
18:07:24.786 > Network lost connection
18:07:27.633 > WiFi disconnected
18:07:27.633 > Try reconnecting
18:07:27.633 > Network lost connection
18:07:30.481 > WiFi disconnected
18:07:30.481 > Try reconnecting
18:07:30.481 > Network lost connection
18:07:33.326 > WiFi disconnected
18:07:33.332 > Try reconnecting

mowegener, 2022-11-12:
Hallo @madmartin @tbnobody ,
ich habe denke mal das Problem gefunden. Ich habe mal im Router einen Gastzugang aufgemacht, dieser hat nur mit dem Board funktioniert, wenn ich das Passwort des Netzwerkes ohne Zahlen eingerichtet habe. In dem Gastzugang habe ich beides ausprobiert, ein Passwort mit und ohne Zahlen. Es hat nur bei dem ohne Zahlen funktioniert. Eigentlich komisch da die Passwörter der Test-Hotspots auch, bzw. nur aus Zahlen bestanden haben und da funktionierte alles einwandfrei. Gibt es da irgendeinen anderen Weg das Passwort einzutragen? Möchte wirklich ungern das Passwort ändern. Beim Gastzugang ist jetzt nur das Problem, dass dieser keinen Zugang zum Netzwerk gibt. Also das Gerät bekommt logischerweise eine IP, aber es kann über diese NICHT aufgerufen werden. Es sollte rein theoretisch funktionieren wenn ich einen AP kaufe, welcher ein separates Netz aufmacht und ein Passwort benutzt welches keine Zahlen enthält. Das wäre aber auch ein wenig umständlich, gibt es nicht evtl. noch eine andere Möglichkeit? Muss ein Passwort welches aus Zahlen und Buchstaben besteht evtl. anders eingegeben werden?

madmartin, 2022-11-12:
@mowegener 
Bitte sei mir nicht böse.... das klingt total verworren was Du da schreibst. Zumindest das letzte...

Zuerst einmal: ja das serial log ist das richtige. Nur ist der von Dir kopierte Teil der uninteressante Teil. Interessant ist was nach dem Reset des ESP Boards passiert. Also bitte das Log aufmachen (Terminalprogramm starten und passend einrichten) und DANN den Reset-Taster vom ESP Board drücken. Du wirst sehen, dann kommt auch was anderes, und DAS wollen wir sehen, bis es sich zu wiederholen beginnt (so wie oben).

Zum Thema WiFi Passwort: Also der WLAN-Name - die SSID - sollte zuerstmal nur aus Ziffern und Buchstaben (ohne Umlaute!) bestehen. Keine Leerzeichen, und nur "harmlose" Sonderzeichen wie Bindestrich oder Unterstrich. Natürlich kann man alles mögliche da noch reinmachen, aber früher oder später findet sich ein Gerät was damit nicht zurechtkommt. 

Das gleiche gilt für das WLAN Passwort. Das darf natürlich Zahlen enthalten! 

Wifi-Kanal 1-11, das hatten wir ja schon besprochen.

mowegener, 2022-11-12:
23:21:16.443 > ets Jul 29 2019 12:21:46
23:21:16.444 >
23:21:16.444 > rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
23:21:16.449 > configsip: 0, SPIWP:0xee
23:21:16.449 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:21:16.455 > mode:DOUT, clock div:2
23:21:16.460 > load:0x3fff0030,len:1184
23:21:16.460 > load:0x40078000,len:13132
23:21:16.465 > load:0x40080400,len:3036
23:21:16.465 > entry 0x400805e4
23:21:17.013 > 
23:21:17.017 > Starting OpenDTU
23:21:17.017 > Initialize FS... done
23:21:17.020 > Reading configuration... done
23:21:17.077 > Initialize Network... done
23:21:17.206 > Setting Hostname... Configuring WiFi STA using new credentials... done
23:21:17.216 > Initialize NTP... done
23:21:17.216 > Initialize MqTT... done
23:21:17.216 > Initialize WebApi... done
23:21:17.226 > Check for default DTU serial... done
23:21:17.226 > Initialize Hoymiles interface... Connection successfull
23:21:17.239 >   Setting radio PA level...
23:21:17.239 >   Setting DTU serial... 
23:21:17.245 >   Setting poll interval...
23:21:17.245 > done
23:21:17.245 > Switch to WiFi mode
23:21:17.248 > Setting Hostname... done
23:21:17.260 > Configuring WiFi STA using existing credentials... E (926) wifi:sta is connecting, return error
23:21:17.273 > [   286][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
23:21:17.279 > done
23:21:17.279 > Configuring WiFi STA static IP... done
23:21:17.376 > WiFi connected
23:21:17.380 > WiFi got ip: 192.168.2.233
23:21:17.384 > Network connected

mowegener, 2022-11-12:
Ich weiß nicht woran dass jetzt liegt, jetzt funktioniert die Verbindung zum WLAN wie aus Geister Hand. Ich schaue aber morgen nochmal rein wie es mit der Verbindung zum WR aussieht. Falls es dann klappen sollte, werde ich das issue schließen. Vielen Dank an alle die aktiv mitgeholfen haben! Ich fühle mich gerade ehrlich gesagt ziemlich doof. Ich habe wirklich ALLES probiert. Das Passwort habe ich 100te male eingegeben und jetzt funktioniert es einfach so. Peinlich!

mowegener, 2022-11-13:
Morgen allerseits!
Das ist der neue Stand was OpenDTU ausspuckt nachdem ich es neu gestartet habe via reset. Habe zwar in anderen Issues ähnliche Probleme gefunden, welche mir aber nicht wirklich weitergeholfen haben. Ich weiß nicht ob irgendjemand hiermit etwas anfangen kann, wäre toll wenn ja. Entschuldigung das ich hier immer noch mit Problemen ankomme. Bin froh dass ich jetzt erstmal schon soweit gekommen bin.
09:42:35.912 > rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
09:42:35.912 > configsip: 0, SPIWP:0xee
09:42:35.912 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:42:35.912 > mode:DOUT, clock div:2
09:42:35.912 > load:0x3fff0030,len:1184
09:42:35.912 > load:0x40078000,len:13132
09:42:35.912 > load:0x40080400,len:3036
09:42:35.912 > entry 0x400805e4
09:42:35.912 >
09:42:35.912 > Starting OpenDTU
09:42:35.912 > Initialize FS... done
09:42:35.912 > Reading configuration... done
09:42:35.912 > Initialize Network... done
09:42:35.989 > Setting Hostname... Configuring WiFi STA using new credentials... done
09:42:36.000 > Initialize NTP... done
09:42:36.000 > Initialize MqTT... done
09:42:36.000 > Initialize WebApi... done
09:42:36.010 > Check for default DTU serial... done
09:42:36.010 > Initialize Hoymiles interface... Connection successfull
09:42:36.024 >   Setting radio PA level...
09:42:36.024 >   Setting DTU serial... 
09:42:36.030 >   Setting poll interval...
09:42:36.030 >   Adding inverter: 116180205245 - PV01 done
09:42:36.035 > done
09:42:36.035 > Switch to WiFi mode
09:42:36.035 > Setting Hostname... done
09:42:36.107 > Configuring WiFi STA using existing credentials... done
09:42:36.124 > Configuring WiFi STA static IP... done
09:42:36.129 > WiFi connected
09:42:36.133 > WiFi got ip: 192.168.2.233
09:42:36.137 > Network connected
09:42:40.750 > Fetch inverter: 116180205245
09:42:40.770 > Request SystemConfigPara
09:42:45.787 > Fetch inverter: 116180205245
09:42:45.787 > Request SystemConfigPara
09:42:45.787 > TX RealTimeRunData Channel: 23 --> 15 80 20 52 45 80 11 24 36 80 0B 00 63 70 9E BB 00 00 00 00 00 00 00 00 F0 AE C2
09:42:46.092 > RX Period End
09:42:46.098 > All missing
09:42:46.098 > Nothing received, resend whole request
09:42:46.103 > TX RealTimeRunData Channel: 40 --> 15 80 20 52 45 80 11 24 36 80 0B 00 63 70 9E BB 00 00 00 00 00 00 00 00 F0 AE C2
09:42:46.402 > RX Period End
09:42:46.407 > All missing
09:42:46.407 > Nothing received, resend whole request
09:42:46.413 > TX RealTimeRunData Channel: 61 --> 15 80 20 52 45 80 11 24 36 80 0B 00 63 70 9E BB 00 00 00 00 00 00 00 00 F0 AE C2
09:42:46.710 > RX Period End
09:42:46.717 > All missing
09:42:46.717 > Nothing received, resend whole request
09:42:46.722 > TX RealTimeRunData Channel: 75 --> 15 80 20 52 45 80 11 24 36 80 0B 00 63 70 9E BB 00 00 00 00 00 00 00 00 F0 AE C2
09:42:47.025 > RX Period End
09:42:47.025 > All missing
09:42:47.025 > Nothing received, resend whole request
09:42:47.032 > TX RealTimeRunData Channel: 3 --> 15 80 20 52 45 80 11 24 36 80 0B 00 63 70 9E BB 00 00 00 00 00 00 00 00 F0 AE C2
09:42:47.334 > RX Period End
09:42:47.334 > All missing
09:42:47.334 > Nothing received, resend count exeeded
09:42:47.339 > TX AlarmData Channel: 23 --> 15 80 20 52 45 80 11 24 36 80 11 00 63 70 9E BB 00 00 00 00 00 00 00 00 2A B5 19 
09:42:48.038 > RX Period End
09:42:48.043 > All missing
09:42:48.043 > Nothing received, resend whole request
09:42:48.049 > TX AlarmData Channel: 40 --> 15 80 20 52 45 80 11 24 36 80 11 00 63 70 9E BB 00 00 00 00 00 00 00 00 2A B5 19
09:42:48.746 > RX Period End
09:42:48.751 > All missing
09:42:48.751 > Nothing received, resend whole request
09:42:48.758 > TX AlarmData Channel: 61 --> 15 80 20 52 45 80 11 24 36 80 11 00 63 70 9E BB 00 00 00 00 00 00 00 00 2A B5 19
09:42:49.456 > RX Period End
09:42:49.462 > All missing
09:42:49.462 > Nothing received, resend whole request
09:42:49.467 > TX AlarmData Channel: 75 --> 15 80 20 52 45 80 11 24 36 80 11 00 63 70 9E BB 00 00 00 00 00 00 00 00 2A B5 19 
[... 126 lines cut ...]
09:43:10.174 > Nothing received, resend whole request
09:43:10.179 > TX AlarmData Channel: 3 --> 15 80 20 52 45 80 11 24 36 80 11 00 63 70 9E CF 00 00 00 00 00 00 00 00 28 E2 38
09:43:10.877 > RX Period End
09:43:10.882 > All missing
09:43:10.882 > Nothing received, resend count exeeded
09:43:10.887 > TX SystemConfigPara Channel: 23 --> 15 80 20 52 45 80 11 24 36 80 05 00 63 70 9E CF 00 00 00 00 00 00 00 00 3C F6 2C
09:43:11.187 > RX Period End
09:43:11.192 > All missing
09:43:11.192 > Nothing received, resend whole request
09:43:11.198 > TX SystemConfigPara Channel: 40 --> 15 80 20 52 45 80 11 24 36 80 05 00 63 70 9E CF 00 00 00 00 00 00 00 00 3C F6 2C
09:43:11.501 > RX Period End
09:43:11.501 > All missing
09:43:11.501 > Nothing received, resend whole request
09:43:11.506 > TX SystemConfigPara Channel: 61 --> 15 80 20 52 45 80 11 24 36 80 05 00 63 70 9E CF 00 00 00 00 00 00 00 00 3C F6 2C
09:43:11.804 > RX Period End
09:43:11.810 > All missing
09:43:11.810 > Nothing received, resend whole request
09:43:11.815 > TX SystemConfigPara Channel: 75 --> 15 80 20 52 45 80 11 24 36 80 05 00 63 70 9E CF 00 00 00 00 00 00 00 00 3C F6 2C
09:43:12.113 > RX Period End
09:43:12.119 > All missing
09:43:12.119 > Nothing received, resend whole request
09:43:12.124 > TX SystemConfigPara Channel: 3 --> 15 80 20 52 45 80 11 24 36 80 05 00 63 70 9E CF 00 00 00 00 00 00 00 00 3C F6 2C
09:43:12.422 > RX Period End
09:43:12.428 > All missing
09:43:12.428 > Nothing received, resend count exeeded

tbnobody, 2022-11-14:
Die manuelle Synchronisierung muss man nur händisch machen wenn der NTP Server nicht erreichbar ist. spät. 1min nach dem Start sollte unter Info --> NTP ein "synced" erscheinen. Dort sollte auch die korrekte Zeit dargestellt werden.

(Unabhängig davon, was wäre bei deinem Screenshot die korrekte Zeit gewesen? 15:08 oder 16:08?)

Die nicht vorhandenen RX zeilen (bzw. die vorhandenen RX timeouts) siehe in deiner vorletzten Nachricht haben nichts mit der Zeit zu tun. Wenn die Zeit nicht passen würde, würden auch keine TX Nachrichten erscheinen.
Dein Problem kann jetzt an vielerlei Dingen liegen:
* Seriennummer ist korrekt?
* IRQ Pin ist korrekt angeschlossen? (Den IRQ Pin kann man per Software nicht prüfen. Alle anderen Pins können überprüft werden indem man auf Info --> System, ganz unten die Radio Infos betrachtet)
* Unter Settings --> DTU ist das PA Level auf Low gestellt? (Für höhere Werte benötigt man ggf. zusätzliche Kondensatoren etc. Mit Low kommt man schon ziemlich weit)

mowegener, 2022-11-14:
Die Current local war hier die korrekte, also 16:08. An den DTU Settings konnte ich bisher nichts ändern da bei mir diese Meldung kommt:
[image]

tbnobody, 2022-11-14:
Dieser Fehler wurde bereits vor 20 Tagen in 9c7dc681b1d91e58a80ccf268d56e3e335e04ee0 behoben... Aber Low sollte (ohne zusätzliche Hardware) sowieso passen.

mowegener, 2022-11-14:
Tatsächlich habe ich die Pins für GPIO4 und GPIO16 jeweils auf dem darunterliegenden Pin gesteckt🤦‍♂️. Schau dann morgen nochmal rein. Panels haben keine Leistung mehr, also WR offline, das typische halt.

mowegener, 2022-11-14:
Kriege aktuell zwar immernoch die "Fehlercodes" von gestern, schau aber mal morgen was passiert.

mowegener, 2022-11-14:
Ich weiß nicht ob ich zu blöd bin, aber wo finde ich bitteschön die .bin Datei, welche ich fürs Update hochladen soll?

mowegener, 2022-11-14:
@tbnobody Achso dankeschön, erinnere mich wieder. War auf einer komplett falschen Seite.🤦‍♂️

stefan123t, 2022-11-15:
> Den IRQ Pin kann man per Software nicht prüfen

@tbnobody man kann doch neben dem RX IRQ auch einen TX IRQ setzen. 
Wäre es nicht möglich etwas zu senden und auf dem konfigurierten Pin einige (Milli-)Sekunden zu warten ?
Wenn keine Rückmeldung TX done kommt, dann stimmt was nicht und man könnte den Timeout für eine Fehlermeldung nutzen.
So könnte man m.E. einige Pins durch probieren bzw. auf Funktion prüfen, oder ?

mowegener, 2022-11-15:
Vielen Dank an alle die mitgeholfen haben. OpenDTU funktioniert mittlerweile einwandfrei und liefert alle Daten. Baue mittlerweile ein OpenDTU für einen Bekannten, hoffe, dass da wenigstens alles korrekt läuft. Wieso ich diese Probleme hatte, kann ich nichts sagen. Genauso wenig kann ich verstehen, wie ich nach etlichem Überprüfen der Pins übersehen konnte, dass zwei wichtige Pins falsch waren. Trotzdem danke an alle die mir auf dem Weg zum Ziel geholfen haben! Das Projekt ist echt spitze und war genau das, was ich gesucht habe. Werde das Programm auf jeden fall weiteremfehlen. Danke an alle die dieses Projekt vorantreiben und damit anderen große Kosten ersparen!

closed 2022-11-15 by mowegener