tbnobody/OpenDTU #274: Hinzufügen einer API

Leviathan09, 2022-10-27:
### Describe the solution you'd like

Gibt es eine Möglichkeit eine API ainzubauen damit evtl. auch Geräte die kein MQTT können per einfachem HTTP Aufruf Befehle senden können?
z.B.: 192.168.x.x/cmd/limit_nonpersistent_relative=50 oder 192.168.x.x/cmd/restart

tbnobody, 2022-10-27:
Würde an dieser Stelle einfach auf die bestehende API verweisen: https://github.com/tbnobody/OpenDTU/pull/233

Leviathan09, 2022-10-27:
> Würde an dieser Stelle einfach auf die bestehende API verweisen: #233

Oh man, wie blöd von mir
Danke

closed 2022-10-27 by Leviathan09

Leviathan09, 2022-10-28:
Hey @tbnobody hab mir das mal angeschaut aber werde nicht schlau draus bzw bekomme es nicht zum laufen.

Kannst du mir evtl anhand des "limit_nonpersistent_relative" ein Beispiel geben wie ich das bewerkstelligen kann?

Die GET Befehle sind kein Problem da es einfach über ein curl geht.
Aber bei POST mit json hört leider mein Verständnis auf.

tbnobody, 2022-10-28:
Versuche es doch mal nach folgender Art und Weise:
`curl -X POST http://192.168.20.110/api/limit/config -d 'data={"serial":"3434", "limit_type":1, "limit_value":100}'`

Leviathan09, 2022-10-29:
Also egal wie ich es versuche, ich bekomme immer gesagt "No value found"
Habe mir jetzt mal versucht mit VSCode deinen Code anzuschauen und wenn ich das richtig verstehe fehlt ihm immer der "Data" Parameter.

Wenn ich das so machen wie in deinem Beispiel kommt der Fehler, wenn ich das Wort data weg lasse kommt er auch. Also ich kapier das nicht.

Dachte mir der API komme ich besser klar als mit MQTT weil da das senden nur über ein extra MQTT Client funktioniert aber nicht aus meinem iobroker heraus.
Aber bekomme beides nicht zum laufen...bin zu doof dafür.

wib100, 2022-10-30:
Hab das gerade mit Postman probiert. Funktioniert wunderbar.
Wichtig ist aber, dass du den String genau so schickt wie von @tbnobody beschrieben! 
Ich habe bei Postman folgenden Body an /api/dtu/config geschickt (POST):
data={"dtu_serial": "19998026187","dtu_pollinterval": 15,"dtu_palevel": 1}

Wie gesagt, ein String in einer Zeile!

Lucifor1976, 2022-10-30:
> Also egal wie ich es versuche, ich bekomme immer gesagt "No value found" Habe mir jetzt mal versucht mit VSCode deinen Code anzuschauen und wenn ich das richtig verstehe fehlt ihm immer der "Data" [...]

Hey, so muss das aussehen: ( GROSGESCHRIBENES DURCH DEINE DATEN ERSETZEN!)
curl -X POST http://OPENDTUIP/api/limit/config -d 'data={"serial":"SERIENNUMMER","limit_type":1,"limit_value":100}'

Leviathan09, 2022-10-30:
> Hey, so muss das aussehen: ( GROSGESCHRIBENES DURCH DEINE DATEN ERSETZEN!) curl -X POST http://OPENDTUIP/api/limit/config -d 'data={"serial":"SERIENNUMMER","limit_type":1,"limit_value":100}'

Also ich hab jetzt deine Zeile kopiert und meine IP sowie Seriennummer eingefügt und es funktioniert.
Warum das vorher nicht ging, keine Ahnung, evtl Copy&Paste Fehler oder so