tbnobody/OpenDTU #283: Error building OpenDTU in vscode

tbnobody, 2022-10-30:
### Discussed in https://github.com/tbnobody/OpenDTU/discussions/282

<div type='discussions-op-text'>

<sup>Originally posted by **joachim-beck** October 30, 2022</sup>
Hello,
i try to build OpenDTU with vscode and followed the instructions given this project.
If I try to build OpenDTU i get the message below.
I am very thankful for any advice.
Thank you for your support.
Joachim

Processing generic (board: esp32dev; framework: arduino; platform: espressif32@>=5.2.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
"C:\Users\jjbec\.platformio\penv\Scripts\python.exe" -m pip install dulwich --global-option="--pure"
WARNING: Implying --no-binary=:all: due to the presence of --build-option / --global-option / --install-option. Consider using --config-settings for more flexibility.
DEPRECATION: --no-binary currently disables reading from the cache of locally built wheels. In the future --no-binary will not influence the wheel cache. pip 23.1 will enforce this behaviour change. A possible replacement is to use the --no-cache-dir option. You can use the flag --use-feature=no-binary-enable-wheel-cache to test the upcoming behaviour. Discussion can be found at https://github.com/pypa/pip/issues/11453
Collecting dulwich
  Using cached dulwich-0.20.50.tar.gz (430 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: urllib3>=1.25 in c:\users\jjbec\.platformio\penv\lib\site-packages (from dulwich) (1.26.12)
Installing collected packages: dulwich
  DEPRECATION: dulwich is being installed using the legacy 'setup.py install' method, because the '--no-binary' option was enabled for it and this currently disables local wheel building for projects that don't have a 'pyproject.toml' file. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/11451
  Running setup.py install for dulwich: started
  Running setup.py install for dulwich: finished with status 'error'
  **error: subprocess-exited-with-error**
  
  × Running setup.py install for dulwich did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help
      
      error: option --pure not recognized
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
**error: legacy-install-failure**

× Encountered error while trying to install package.
╰─> dulwich

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
*** Error 1
ModuleNotFoundError: No module named 'dulwich':
  File "C:\Users\jjbec\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\jjbec\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\jjbec\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\jjbec\OneDrive\Dokumente\Solar\OpenDTU\auto_firmware_version.py", line 16:
    from dulwich import porcelain
============================================================== [FAILED] Took 2.93 seconds ==============================================================
Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:02.930
======================================================== 1 failed, 0 succeeded in 00:00:02.930 ======================================================== 

 *  Der Terminalprozess "C:\Users\jjbec\.platformio\penv\Scripts\platformio.exe 'run'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 
</div>

tbnobody, 2022-10-30:
I was able to reproduce this issue. Let me just doublecheck why it occours.

closed 2022-10-30 by tbnobody

Marc--, 2022-10-30:
Hi,
I have the same problem, i was trying to compile the lastest version to update my box. 
I'll wait for your answer ! 

Thanks for the job done.

tbnobody, 2022-10-30:
Fixed in current master...

commit 6a3fbc7a (2022-10-30): Fix #283: Fix python error when building firmware