tbnobody/OpenDTU #294: Wifi client runs out of processes - Olimex hardware

stromibaer, 2022-11-02:
### What happened?

Starting ESP and it runs and after a while i get following error messages on serial port:

IP communication is down and ESP is not reachable via webinterface. 
MqTT is also down.
Hoymiles communication is running.

### To Reproduce Bug

After reboot, it is running and will get same error again.

### Expected Behavior

1c3df51 was running fine. 
Is this bug special for Olimex hardware?

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

b6295e7

### Relevant log/trace output

```Shell
[3600895][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3601895][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3602895][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3603895][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3604899][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3605226][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3605899][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3606899][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3607431][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3608405][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
[3609410][E][WiFiClient.cpp:422] write(): fail on fd 48, errno: 11, "No more processes"
```

tbnobody, 2022-11-02:
Both mentioned git hashes are referring to commits which do not belong to this repository. I am not sure what's changed in this foreign repos.

stromibaer, 2022-11-02:
Ok, will try it with ESP32 generic hardware to reproduce error maybe. I have some other troubles with the Olimex POE hardware too.

tbnobody, 2022-11-02:
You could also try if this issue occurs with a unmodified source. Because the issue looks like that there are too many messages in a send queue. But I don't see this issue in a other installation even with multiple days of runtime

rosch99, 2022-11-02:
Just for info, i am running OpenDTU Olimex since summer with no issues. I am only rebooting for updates. 

BIG thanks to @tbnobody for the great software!

stromibaer, 2022-11-03:
I think my Olimex has some hardware troubles, will send it back. Same image is running on ESP32 default hardware without issues.

stromibaer, 2022-11-04:
Looks like the ESP is running out of memory. There is already a discusion at
[https://github.com/espressif/arduino-esp32/issues/6129](url) 
But i do not know how to fix the code and save memory. Has somebody an idea?
[https://github.com/stromibaer/OpenDTU-Victron/tree/master/src](url)
Most code is in file [https://github.com/stromibaer/OpenDTU-Victron/blob/master/src/MqttVictronPublishing.cpp](url)
In general code is working fine, but code quality is maybe not the best one....

closed 2022-11-19 by stromibaer