tbnobody/OpenDTU #298: Passwort für Webinterface

tm-107, 2022-11-03:
### Describe the solution you'd like

Ein simpler Passwortschutz (basic-auth) wie bei Tasmota wäre aus meiner Sicht wünschenswert, damit nicht jeder im Heimnetz die Einstellungen ändern kann.
Außerdem könnte man dann das Webinterface auch aus dem WWW erreichbar machen, sofern einem basic-auth als Schutz ausreicht.

tbnobody, 2022-11-03:
Eine Authentifizierung ist schon auf meiner Liste. 
Aber bitte, bitte, niemals nie ein Gerät/Webseite ohne SSL public ins Internet stellen. Da SSL auf einem ESP32 eher kompliziert wird empfehle ich da eher einen VPN Server am Router zu nutzen. (vgl. Wireguard auf der Fritzbox)

tm-107, 2022-11-03:
Danke für die Rückmeldung!
Mit dem SSL hast Du schon recht. Aber da OpenDTU und einige Tasmotas bei mir in einem separaten WLAN-Netz hängen (mit eigenem Subnetz im LAN), halte ich die Gefahr in meinem Fall für vertretbar.
Ein VPN (OpenVPN über einen RaspberryPi) existiert ebenfalls. Das nützt mir allerdings nichts wenn ich auf Arbeit bin, da ich dort keinen VPN-Client installieren/nutzen kann. Und die Wartung meiner Home-IT erledige ich zu 95% von Arbeit aus ...

D3R-ST3FAN, 2022-11-03:
Authentifizierung find ich auch eine gute Idee, auch (erst mal) ohne SSL.

D3R-ST3FAN, 2022-11-03:
> Danke für die Rückmeldung! Mit dem SSL hast Du schon recht. Aber da OpenDTU und einige Tasmotas bei mir in einem separaten WLAN-Netz hängen (mit eigenem Subnetz im LAN), halte ich die Gefahr in [...]

Ich hab das über einen Reverse Proxy realisiert. Der terminiert nebenbei auch noch die SSL Verbindung.

tbnobody, 2022-11-03:
> Ich hab das über einen Reverse Proxy realisiert. Der terminiert nebenbei auch noch die SSL Verbindung.

Sowas ist auch noch i.O. da kann man ja auch zusätzlich noch nen Basic auth einbauen damit man nicht direkt auf dem Gerät landet....

tm-107, 2022-11-03:
> Ich hab das über einen Reverse Proxy realisiert. Der terminiert nebenbei auch noch die SSL Verbindung.

Auf die Idee bin ich leider nicht selbst gekommen.
Aber jetzt fungiert mein (sowieso laufender) Apache auch als Reverse Proxy mit integriertem basic-auth :-)

D3R-ST3FAN, 2022-11-04:
> Auf die Idee bin ich leider nicht selbst gekommen. Aber jetzt fungiert mein (sowieso laufender) Apache auch als Reverse Proxy mit SSL-Anbindung und integriertem basic-auth :-)

Ich nutze HAProxy, der läuft bei mit auf der OPNSense Firewall  (die quasi Router missbraucht wird).

Im selben (V)LAN Segment fänd ich Authentifizierung direkt an OPENDTU dann halt noch besser.
Hab ne Menge I-OT Kram und man weiß ja nie.

tm-107, 2022-11-04:
> Im selben (V)LAN Segment fänd ich Authentifizierung direkt an OPENDTU dann halt noch besser. Hab ne Menge I-OT Kram und man weiß ja nie.

Sehe ich genauso ...

hubsi5, 2022-11-08:
Wie lautet nun das Standardpasswort der Oberfläche?

D3R-ST3FAN, 2022-11-08:
Passwortschutz ist ja jetzt eingebaut, SUPER!
Ich denke aber, der sollte ein/ausschaltbar sein oder?

distanzcheck, 2022-11-08:
Wie lautet denn der USername und das Paswort ?

tbnobody, 2022-11-08:
It's documented in the README
https://github.com/tbnobody/OpenDTU/blob/11c272b1fbc7b5616cc464318fbe32ba9f401d08/README.md?plain=1#L171

tm-107, 2022-11-08:
Nur die Settings mit Passwort fände ich erst mal okay. Vielen Dank dafür!
Aber über die "Live Data"-Seite kann ich dennoch ohne Passwort das Inverter-Limit setzen/ändern oder den Inverter komplett an-/abschalten ...
Also entweder diese beiden Punkte wandern ins Settings-Menü oder der Passwortschutz gilt (optional?) für das komplette WebUI. Anders macht es in meinen Augen keinen Sinn.

Mein zusätzlicher Passwortschutz über den Reverse-Proxy bleibt also vorerst bestehen :-)

tbnobody, 2022-11-08:
> Aber über die "Live Data"-Seite kann ich dennoch ohne Passwort das Inverter-Limit setzen/ändern oder den Inverter komplett an-/abschalten ...

Siehe #318 ist bekannt. War aber etwas komplizierter als der Rest. Daher wollte ich erstmal das bestehende so publizieren

dachauan, 2022-11-09:
Leider kann ich mich nach dem Update nicht anmelden, User: admin Pwd: openDTU42.
Fehlermeldung: Invalid credentials
Stimmt das default Pwd?

tbnobody, 2022-11-09:
Das Passwort ist das gleiche was unter Settings --> Security eingetragen wurde. Also das AP Passwort. Wenn du das jemals geändert hast, dann bitte dieses verwenden.

henne49, 2022-11-09:
Ich habe das gleiche Problem mit dem Passwort. Ich habe das Update per OTA eingespielt.

tbnobody, 2022-11-09:
Du verwendest auch wirklich "admin" als Benutzer und "openDTU42" als Passwort?
Wenn du den ESP in dein WLAN gebracht hast, dann hast du dich ja das 1. mal auf den integrierten Access Point verbunden und dort die Einstellungen vorgenommen. Welches Passwort hast du da verwendet? Wenn du den ESP kurz vom Strom nimmst und wieder ansteckst geht dieser Access Point wieder auf. Kannst du dich noch verbinden?

henne49, 2022-11-09:
ja ich habe extra per Copy und Paste gearbeitet. Auch nach Reboot Stromlos komme ich zwar drauf, aber die Credentials funktionieren nicht.

tbnobody, 2022-11-09:
Lade mal das WebUI vorsichtshalber mit STRG+F5 neu. Und ich Frage nochmals, weil da noch nicht direkt eine Antwort kam :)
das Passwort das du beim Access Point eingegeben hast war "openDTU42"?

henne49, 2022-11-09:
ja ich habe "openDTU42" ohne Anführungszeichen genutzt beim Login ging nicht, ich habe jetzt auch neu geflasht und kann nicht mehr im AP Modus per WLAN connecten, WPA2-PSK key openDTU42. Test vom Handy diesmal. Ich habe auch die defaults.h überprüft:

#define ACCESS_POINT_NAME "OpenDTU-"
#define ACCESS_POINT_PASSWORD "openDTU42"
#define AUTH_USERNAME "admin"

muss ich den Flash nochmals vorher erasen, wenn ich per platform.io aus VS Code arbeite?

tbnobody, 2022-11-09:
> muss ich den Flash nochmals vorher erasen, wenn ich per platform.io aus VS Code arbeite? Damit ich ein Factory reset mache?

Ja, weil beim normalen Flashen die Partition mit der Config nicht überschrieben wird.

henne49, 2022-11-09:
Danke, jetzt läuft alles wieder. Tollen Job den ihr hier macht! Danke dafür, funktioniert mega!

m0tschi, 2022-11-11:
Kleiner Hinweis am Rande, wenn man im AP kein Passwort hinterlegt hat, sperrt man sich aus :)
Trotzdem wichtige Verbesserung! 💯 
Danke für den Tipp mit erst Erease und dann erst Flashen 👍

tbnobody, 2022-11-11:
Darf ich fragen wie du es geschafft hast kein Passwort zu hinterlegen? Eigentlich sollte da eine Validierung drin sein dass das Passwort mindestens 8 Zeichen lang sein muss.

dachauan, 2022-11-11:
Wie bei mir, hatte auch kein Passwort hinterlegt, openDTU42 ging trotzdem nicht. Hab den ESP platt gemacht und neu installiert.

hubsi5, 2022-11-11:
Merkwürdig.
Bei mir war auch nie etwas hinterlegt, aber nach de flashen ging ganz normal das hinterlegte Standardpasswort.

tbnobody, 2022-11-11:
@dachauan bzw. @m0tschi wisst ihr noch von welcher Version ihr ungefähr upgedated  habt?

dachauan, 2022-11-11:
Ich hab zur Zeit jeden Tag upgedatet, genaue Version kann ich dir nicht sagen. Erstinstallation war am 28.10.

m0tschi, 2022-11-12:
@tbnobody 
> Darf ich fragen wie du es geschafft hast kein Passwort zu hinterlegen? Eigentlich sollte da eine Validierung drin sein dass das Passwort mindestens 8 Zeichen lang sein muss.

hatte das in der defaults.h geändert beim ersten Flashen... bin also selbst Schuld.
die Version war vom Vortag selbst kompiliert.

madmartin, 2022-11-15:
@tbnobody  
Dieses "Issue" Passwortschutz für Webinterface ist doch jetzt erledigt und könnte geschlossen werden?

tm-107, 2022-11-15:
Sehe ich nicht so. Es steht ja noch im Raum, wahlweise das komplette WebUI per Passwort zu schützen.
Wenn das Ganze aus dem WWW erreichbar sein soll, möchte ich nicht dass jeder darauf zugreifen kann und anhand der Info/Network-Seite auch noch einige Infos zu meinem Home-LAN erhält.

tbnobody, 2022-11-15:
Das ist der Grund warum es noch offen ist :)

fritzCARaldo, 2022-11-19:
Hallo! Ich bin neu hier und habe mir meinen OpenDTU basteln lassen. Habe soeben erfolgreich auf OpenDTU Build #359: Commit [854af46] geupdatet. Dort wird jetzt ein LogIn abgefragt. Ich bin wahrscheinlich zu Grün: Was sind denn hier die Grunddaten für den LogIn? Danke!

madmartin, 2022-11-19:
> Was sind denn hier die Grunddaten für den LogIn? Danke!

Username ist "admin" und das Passwort dasselbe wie für den Accesspoint - standard ist "openDTU42" und Du solltest das dann ändern.

tbnobody, 2022-11-25:
Alle API Endpunkte können nun (optional) Passwortgeschützt werden. Würde den Issue entsprechend schließen.

closed 2022-11-25 by tbnobody