tbnobody/OpenDTU #318: Inverter Limit can be changed without being logged in first / Password protection does not work

D3R-ST3FAN, 2022-11-08:
### What happened?

With commit 11c272b inverter limit can be changed without being logged in in WebUI.
Inverter can also be restarted and trurned on/off. 

These functions should be protected as well.

### To Reproduce Bug

go to status page and set limit

### Expected Behavior

changing the limit should not be possible if not logged in

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

11c272b

tbnobody, 2022-11-08:
Is on my list... Will follow the next days... Also Power On/Off/Restart will be protected...

tbnobody, 2022-11-08:
Should be fixed with e0b069ff3da978c370157996870096e738e392ab

closed 2022-11-08 by tbnobody