tbnobody/OpenDTU #320: [Request] DTU Settings (Poll Interval) morgens automatisch neu setzen?

D3R-ST3FAN, 2022-11-08:
### Is your feature request related to a problem? Please describe.

Ich habe den Poll Intervall angepasst.
Scheinbar wird beim morgentlichen Inverterstart diese Poll Intervall nicht übernommen.

### Describe the solution you'd like

Beim Inverterstart Poll Interval auf den letzten Wert setzen.

tbnobody, 2022-11-08:
Wie kommst du darauf? Ist es im Webinterface zurückgesetzt?

D3R-ST3FAN, 2022-11-08:
> Wie kommst du darauf? Ist es im Webinterface zurückgesetzt?

Nein im WebIF steht der "alte" Wert.
Vielleicht interpretiere ich das auch falsch. 
Nach dem morgentlichen Neustart zählt Data Age immer bis auf 15 Sekunden (default?) hoch.

tbnobody, 2022-11-08:
Nur damit ich es korrekt verstehe. Der Default Wert in den Einstellungen ist ja 5 Sek. Du setzt den Wert z.B. auf 10Sek und in der früh steht er wieder auf 5 Sekunden?

Das Hochzählen kann unabhängig davon verschiedenste Gründe haben. Die Zeit gibt an, dass zwischen verschiedenen Polls einfach 5 Sek. gewartet wird. Aber in so einem 5 Sek. Rhythmus können verschiedenste Pakete mit unterschiedlichen Timeouts reinfallen. Also z.B. ein Statistikpaket, ein Paket zum Auslesen des Eventlog, oder zum Lesen des aktuellen Limits. Was genau passiert sieht man am genauesten auf der Seriellen Konsole.

D3R-ST3FAN, 2022-11-10:
> Nur damit ich es korrekt verstehe. Der Default Wert in den Einstellungen ist ja 5 Sek. Du setzt den Wert z.B. auf 10Sek und in der früh steht er wieder auf 5 Sekunden?

nein, der eingestellte Wert bleibt erhalten.

Ich nahm bisher an, dass dieser Poll Intervall an den WR übertragen wird und der dann alle X Sekunden die Daten sendet.
Meine Beobachtung war, dass nach einer Änderung des Polling Intervalls, auch der Data Age Zähler entsprechend hochzählt und bei erreichen von Polling Intervall neu startet. Nach dem Inverter Neustart am Morgen bleibt Polling Intervall in openDTU erhalten, aber der Counter geht bis 15.

Ich werde das aber bei nächster Gelegenheit noch mal prüfen.

tbnobody, 2022-11-10:
Das poll intervall wird nicht an den inverter übertragen. der inverter sendet von sich aus garnichts. man muss alle werte separat anfragen (also statistik, device info, aktuelle limits etc). die zeit zwischen den anfragen ist einfach dieser wert. 
Siehe hier: https://github.com/tbnobody/OpenDTU/blob/8b158a029e94cdf7b44b1404e60b6c860f7f3dd3/lib/Hoymiles/src/Hoymiles.cpp#L21

indeed1, 2022-11-11:
Ich habe gestern testweise mein Intervall auf einen anderen als den Defaultwert gesetzt. Heute nach dem Inverterstart ist weiterhin der von mir gesetzte Wert sowohl wirksam als auch in den Einstellungen sichtbar. Insofern kann ich da kein Problem erkennen - oder ich verstehe es nicht.

closed 2022-11-14 by D3R-ST3FAN