tbnobody/OpenDTU #322: Passwortschutz scheint allergisch gegen Sonderzeichen - Aussperrgefahr!

indeed1, 2022-11-09:
Hallo,

zunächst mal herzlichen Dank für dies großartige Projekt! Tolle Leistung! Ich verfolge es schon eine Weile und hatte dank der vorbildlichen Dokumentation auch keine Probleme, mein eigenes OpenDTU in Betrieb zu nehmen, es läuft reibungslos.

Aber nachdem jetzt Teile der Weboberfläche mit einem Passwortschutz versehen sind, habe ich mich durch Einrichtung eines eigenen Passworts ausgesperrt. Ausgehend von Gewohnheiten mit Tasmota habe ich arglos ein Passwort gewählt, in dem auch Sonderzeichen vorkommen (vorliegend ein Umlaut und "ß"). Das wird zwar bei der Einrichtung klaglos angenommen, aber bei späteren Login-Versuchen dennoch nicht als gültig erkannt. Das ist also eine potenzielle "Falle", der am Ende vermutlich nur durch Löschen und Neu-Flashen zu entkommen ist. Vielleicht lässt sich das irgendwie entschärfen?

Und noch eine andere Überlegung: Ist die offene Anzeige der Inverter-Seriennummer(n) auf der Live-Seite wirklich nötig? Aus meiner Sicht sollten die Seriennummern wie Passwörter behandelt werden, denn sie sind faktisch das einzige "Geheimnis", welches die Inverter-Funkschnittstelle gegen unbefugten Zugriffe sichert! Der neue Passwortschutz der Web-Oberfläche, obwohl gewiss eine gute Idee, kann das ja nicht wirklich, da mit separater Hardware umgehbar. 
Zugegeben, so ein Angriff ist ein unwahrscheinliches Szenario, und der potenzielle Schaden zudem gering, aber aus grundsätzlichen Überlegungen würde ich dennoch dafür plädieren, die Anzeige der Seriennummer aus der Live-Seite rauszunehmen, so wie man ja auch ein Passwort niemals dort anzeigen würde. Zusätzliches Problem: anders als ein Passwort ist sie nicht änderbar, d.h. wenn sie einmal in falsche Hände gekommen ist, ist das nie mehr heilbar.

Grüße, Manfred

tbnobody, 2022-11-09:
Das mit den Sonderzeichen schaue ich mir an. Das darf so nicht sein. (Weißt du um welches Sonderzeichen es sich dabei genau handelte?)

Bzgl. der Seriennummer. Der ESP sollte so und so NIEMALS NIE public im Internet zu finden sein. Es wird alles ohne SSL übertragen (das ist am ESP eine nicht ganz triviale Angelegenheit). Hier ist auf jeden Fall zu einem VPN Zugang (z.B. via Fritzbox) zu raten!
Die Seite mit der Seriennummer ist also sowieso nur nur von vertrauenswürdigen Personen zu erreichen
Sollte sich ein Angreifer im eigenen WLAN oder VPN aufhalten hat man ggf. ein anderes Problem.
Es würde zumal auch nichts bringen die Seriennummer nicht anzuzeigen, da sie intern bei vielen Anfragen wie Eventlog lesen, Limit/Power setzen übertragen werden muss. Ich baue ggf. später noch eine Funktion ein um die gesamte Web UI Passwort zu schützen (optional).

tbnobody, 2022-11-09:
Hm Ok... dadurch das das Passwort auch für die WPA-PSK Authentifzierung verwendet wird dürfen sowieso nur ASCII Zeichen verwendet werden. Das Internet sagt hier u.A. zwischen Code 32 und 126. Dies schließt Umlaute aus.

indeed1, 2022-11-09:
> Weißt du um welches Sonderzeichen es sich dabei genau handelte?
ü und ß

tbnobody, 2022-11-09:
Die Javascript Funktion `window.btoa` mag keine Umlaute usw... Eine Lösung gibt es hier:
https://stackoverflow.com/questions/3626183/javascript-base64-encoding-utf8-string-fails-in-webkit-safari
Dann klappts auch mit den Umlauten und wenn den ESP noch nicht platt gemacht hast solltest du dich nach einem Update auch wieder anmelden können (aber noch ist der code noch nicht commited).

commit 7d1cdb05 (2022-11-09): Fix #322: German umlauts create mistakes when using them in passwords

See https://stackoverflow.com/questions/3626183/javascript-base64-encoding-utf8-string-fails-in-webkit-safari
closed 2022-11-09 by tbnobody