tbnobody/OpenDTU #328: [Request] Sync NTP from Browser if date is Thu Jan 01 00:00:00 1970 UTC

MyIgel, 2022-11-10:
### Is your feature request related to a problem? Please describe.

At the moment the board starts with `Thu Jan 01 00:00:00 1970 UTC` as the start datetime which prevents it with communication with the inverter (some cert errors) before clicking "sync time".

### Describe the solution you'd like

It would be great to automatically sync  / ask to sync the time based on the current set browser time.

### Describe alternatives you've considered

Ignoring the cert checks if the date is totally off or at least use the build datetimestamp on boot

tbnobody, 2022-11-10:
Question, whats so special with your setup that you don't have access to a NTP server?

MyIgel, 2022-11-10:
For example when initially testing it over the hotspot and being confused that it does not work.

ghostridernrw40, 2022-11-23:
Just add a DNS-Server and don't leave it blank. Then it works great.

tbnobody, 2022-11-28:
The Live View shows a very big message when no time is set. It also provides a link which directly leads to the NTP sync page. I don't like the idea of setting the time to a compiled time as it will mess up the eventlog.

closed 2022-11-28 by tbnobody