tbnobody/OpenDTU #332: Keine Verbindung zum Wechselrichter (HM-600)

Ascendor, 2022-11-12:
Hallo, mir gehen die Ideen aus:

Ich habe dieses ESP-Board: https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware (genauer: Olimex ESP32-POE-ISO-EA-IND) mit diesem Funkmodul: https://www.sparkfun.com/products/705 und dieser Antenne: https://www.sparkfun.com/products/558

Ich kriege keine Verbindung. Ich habe versucht, nah an den Wechselrichte ranzugehen (Entfernung 1-2m plus ein Holz-/Bitumen-Dach dazwischen. Habe verschiedene USB-Netzteile probiert (1.8A), verschiedene USB-Kabel. Habe Netzwerk via WLAN und via Ethernet probiert (um Interferenzen zu minimieren). Habe die Sendeleistung erhöht.
Nichts. Die Seriennummer habe ich 10 mal kontrolliert, ist definitiv richtig.

Was kann ich noch tun? Ich las hier von einer seriellen Schnittstelle. Wie kann ich die auslesen? 

danke für jede Hilfe!

tbnobody, 2022-11-12:
NTP Sync ist ok?
Modul wird korrekt erkannt?

Serielle Konsole lesen (putty Laden, auf seriellen Modus umschalten, com Port eintragen, baudrate auf 115200 setzen)

Ascendor, 2022-11-13:
Danke @tbnobody, für die schnelle Antwort.

NTP sync ist OK, denke ich. Wenn ich näher an den Wechselrichter rangehe, verliere ich manchmal die WLAN-Verbindung, weil der AP dann zu weit weg ist. dann mache ich aber einen manuellen Sync. Reicht das auch? 

[image]

Das Modul wird erkannt, ja:

[image]

Konsolenoutput:
`▒▒▒▒=ETH start
done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize MqTT... done
Initialize WebApi... done
Check for default DTU serial... done
Initialize Hoymiles interface... Connection successfull
  Setting radio PA level...
  Setting DTU serial...
  Setting poll interval...
  Adding inverter: 114182120717 - Hm done
done
Switch to WiFi mode
E (5306) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x                                                                                                                                                             3ffb26e0 (config or if_key is NULL or duplicate key)
E (5308) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
[  5319][E][ETH.cpp:270] begin(): esp_eth_mac_new_esp32 failed
Setting Hostname... done
E (5346) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x                                                                                                                                                             3ffb2670 (config or if_key is NULL or duplicate key)
E (5348) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
[  5359][E][ETH.cpp:270] begin(): esp_eth_mac_new_esp32 failed
Configuring WiFi STA using existing credentials... E (11121) wifi:sta is connect                                                                                                                                                             ing, return error
[  5383][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
done
Configuring WiFi STA DHCP IP... done
Fetch inverter: 114182120717
Request SystemConfigPara
WiFi connected
WiFi got ip: 192.168.178.5
                          Network connected
Fetch inverter: 114182120717
Request SystemConfigPara
Fetch inverter: 114182120717
Request SystemConfigPara
Fetch inverter: 114182120717
Request SystemConfigPara
Fetch inverter: 114182120717
Request SystemConfigPara
Fetch inverter: 114182120717
Request SystemConfigPara
Fetch inverter: 114182120717
Request SystemConfigPara
Fetch inverter: 114182120717
Request SystemConfigPara
TX RealTimeRunData Channel: 23 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 2F 00 00 00 00 00 00 00 00 6C 01 69
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 40 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 2F 00 00 00 00 00 00 00 00 6C 01 69
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 61 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 2F 00 00 00 00 00 00 00 00 6C 01 69
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 75 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 2F 00 00 00 00 00 00 00 00 6C 01 69
RX Period End
All missing
[... 737 lines cut ...]
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 61 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 A7 00 00 00 00 00 00 00 00 6A 07 E1
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 75 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 A7 00 00 00 00 00 00 00 00 6A 07 E1
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData Channel: 3 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 A7 00 00 00 00 00 00 00 00 6A 07 E1
RX Period End
All missing
Nothing received, resend count exeeded
TX AlarmData Channel: 23 --> 15 82 12 07 17 80 15 62 60 80 11 00 63 70 B1 A8 00 00 00 00 00 00 00 00 40 5D 84
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 40 --> 15 82 12 07 17 80 15 62 60 80 11 00 63 70 B1 A8 00 00 00 00 00 00 00 00 40 5D 84
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData Channel: 61 --> 15 82 12 07 17 80 15 62 60 80 11 00 63 70 B1 A8 00 00 00 00 00 00 00 00 40 5D 84
R
`

Ascendor, 2022-11-14:
Ach so - sollte ich das hier lieber in Discussions posten als in Issues? Ich sehe dass andere da ihre Verbindungsprobleme posten...

Alfons-Wa, 2022-11-14:
Ich kann nur sagen, dass bei mir von insgesamt vier Modulen lediglich eins, nämlich eines ohne Antenne, störungsfrei funktioniert, seltsamerweise auch über eine Strecke von ca. 7Metern mit einem Ziegeldach dazwischen und das ohne angelötetem 10uf-Kondensator. Mit war der Empfang schlechter. Eines mit Antenne funktionierte nur mit 10uf-Kondensator an den Spannungsanschlüssen, aber auch nicht 100% störungsfrei, eines mit Antenne gar nicht und eines ohne Antenne ebenfalls nur ab und zu. Es scheint da eine Menge an schlechten Modulen auf dem Markt zu sein. Hat mich auch Nerven gekostet.

Ascendor, 2022-11-14:
Puh, das klingt noch nach viel Arbeit. Danke @Alfons-Wa. 
Kannst du mir zu der Kondensatorlösung noch etwas sagen? Wo muss wie welcher Kondensator eingebaut werden?

Alfons-Wa, 2022-11-14:
Direkt an die Pins der Stromversorgung einen 10uf-Kondensator, der nah an der Höhe der Versorgungsspannung sein sollte, anlöten. 10uf/10V sollten reichen. Meine Erfahrungen damit sind allerdings durchwachsen, aber viele berichten, dass die teils schwächlichen Spannungsregler der ESP32-Board besser damit zurechtkommen. Bei einem gut designten ESP32-Board halte ich das aber für überflüssig.

Ascendor, 2022-11-14:
Danke. Bei meinem Board habe ich dann auch wenig Hoffnung, aber wenn es keine weiteren Ideen gibt, werde ich es wohl probieren

tbnobody, 2022-11-14:
Ich habe auch gehört das module von ebyte wohl etwas teurer aber generell zuverlässiger sind. habe es persönlich noch nicht probiert

Ascendor, 2022-11-14:
> Ich habe auch gehört das module von ebyte wohl etwas teurer aber generell zuverlässiger sind. habe es persönlich noch nicht probiert

@tbnobody, siehst Du im Log vom COM-Port noch irgendwelche Hinweise?

tbnobody, 2022-11-14:
Leider nein...
Ich sehe folgendes:
`Initialize Hoymiles interface... Connection successfull`
Sagt das die SPI Verbindung zum NRF  Modul ok ist... IRQ PIN könnte noch falsch verbunden sein

```
TX RealTimeRunData Channel: 23 --> 15 82 12 07 17 80 15 62 60 80 0B 00 63 70 B1 2F 00 00 00 00 00 00 00 00 6C 01 69
RX Period End
All missing
```

Hier wird was zum NRF Modul geschickt. D.h. es wurde eine Zeit gesetzt usw. und Daten werden gepollt. Es kommt nur nichts zurück. Kann entweder an einer falschen Inverter Seriennummer liegen oder wirklich am Modul.

mattfullerton, 2022-11-14:
Ich habe das gleiche Problem. Ich habe jetzt alle Verbindungen mit einem Multimeter geprüft und die 3.3V am Funkmodul gemessen.

10uf-Kondensator sollte über die Stromversorgungspins vom Funkmodul gehen, oder vom ESP-Board?

Habe jetzt 5 weitere Funkmodule bestellt...

Ascendor, 2022-11-14:
Ich habe IRQ noch mal neu verlötet (meine Lötkünste sind... ausbaufähig). Ich habe jetzt zum ersten mal "Interrupt received" im Log gelesen. Kann das überhaupt sein, wenn nachts der Wechselrichter aus ist - kann der Interrupt von anderen Signalen (WLAN) kommen - oder hab ich jetzt eher nen Kurzschluss gebaut?

tbnobody, 2022-11-14:
Wenn du jetzt interrupt received bekommst deutet das eher auf einen Kurzschluss oder auf eine falsche pinbelegung hin :-( 
Von wlan usw kann das eigentlich nicht kommen da zu diesem Zeitpunkt ein empfangenes Paket schon einmal mit einem crc Code im  NRF selbst verifiziert wurde

Ascendor, 2022-11-15:
Oh, ich habe mal einen ersten kleinen Fortschritt gemacht: 

[image]

Seitdem wieder keinerlei Verbindung. Aber immerhin.

stefan123t, 2022-11-15:
@Ascendor @Alfons-Wa auf welche Sendeleistung habt Ihr Eure NRF24 Module eingestellt.
Bei mir reicht bei denen mit externer Antenne LOW oder MIN.

Für die billigen ohne extra Antenne muß ich die Leistung auf HIGH hochdrehen.
Der Elko kommt ggf. direkt an Pin 1 & 2, also an GND & +3.3V des NRF Moduls.
Das dient dazu daß die kleinen Jumperwire teilweise nicht ausreichend dick sind und ggf. nicht die 150mA Strom die der NRF24 beim Senden braucht beibringen können. 
Teilweise gibt es auch Probleme mit dem USB-Steckernetzteil und/oder dem 3.3V Festspannungsregler auf den ESP Boards.  

Manche NRF24 Module haben eine Pinbelegung spiegelverkehrt auf die Unterseite gedruckt.
Das verwirrt zusätzlich, bitte die Pinouts im Internet von oben betrachten, dann passt idR. alles.
D.h. mit Antenne nach rechts und Bauteilen oben, ist der Pin 1 ganz links außen (und oben) GND!
Der hat auch extra ein kleines weißes Quadrat auf der Platinenoberseite aufgedruckt, da bitte zu zählen beginnen. Danke!

Ascendor, 2022-11-15:
@stefan123t: Ich habe alle möglichen Sendeleistungen durchprobiert. Als ich heute früh einmalig Verbindung hatte, war es auf HIGH eingestellt. 

> D.h. mit Antenne nach rechts und Bauteilen oben, ist der Pin 1 ganz links außen (und oben) GND!
Stimmt das für alle Hersteller/Module? Ich habe ja das hier: https://www.sparkfun.com/products/705 und habe aktuell den obersten Pin mit GND verlötet (in der von Dir genannten Ausrichtung). Der ist aber auch glasklar mit GND beschriftet und es gibt nur eine Reihe (kein links/rechts). Und schließlich habe ich oben ein paar Pakete erfolgreich empfangen - könnte das überhaupt sein, wenn es komplett falsch verdrahtet wäre?

Alfons-Wa, 2022-11-15:
In meinem Fall hat die eingestellte Sendeleistung kaum einen bemerkbaren Einfluss. Ich habe sie auf LOW stehen und sehr guten Empfang.

PAMediaService, 2022-11-15:
Hier die Belegung zum PoE

[env:olimex_esp32_poe]
; https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware
board = esp32-poe
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=15
    -DHOYMILES_PIN_MOSI=2
    -DHOYMILES_PIN_SCLK=14
    -DHOYMILES_PIN_IRQ=13
    -DHOYMILES_PIN_CE=16
    -DHOYMILES_PIN_CS=5
    -DOPENDTU_ETHERNET

Ascendor, 2022-11-16:
Ich habe mir nun neue NRF-Module bestellt. Nach ein paar Versuchen, Lötstellen auszubessern, ist die Situation nun bei mir schlechter als vorher (ab und zu wird das Modul nicht mehr als verbunden angezeigt). Ich versuche mein Glück mit neuen/anderen.

Ascendor, 2022-11-18:
Mit den neuen Modulen hat es sofort geklappt. Und die sind auch noch ca. Faktor 8 günstiger... 

Die hier habe ich bestellt: https://www.amazon.de/gp/product/B07VQ838KT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 

Danke für die Hilfe. Ich spende noch eine Kleinigkeit für das Projekt.

mattfullerton, 2023-11-03:
Ich hatte zum Schluss Erfolg mit dem [ICQUANZX Modul](https://www.amazon.de/-/en/gp/product/B07VQ838KT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) - Fazit aus den vielen Versuchen: https://github.com/tbnobody/OpenDTU/discussions/339