tbnobody/OpenDTU #333: [Request] Support for TTGO T-Internet PoE

fromCologne, 2022-11-12:
### Is your feature request related to a problem? Please describe.

I searched for a Eth (not WiFi) solution and a TTGO I-Internet PoE was in my portfolio.
http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1344&FId=t28:21:28

### Describe the solution you'd like

see below

### Describe alternatives you've considered

I changed the software to support TTGO T-Internet-PoE ESP32 module.

platformio.ini
```
[platformio]
default_envs = TTGO_esp32_poe

build_flags =
    -D=${PIOENV}
    -DCOMPONENT_EMBED_FILES=webapp_dist/index.html.gz:webapp_dist/zones.json.gz:webapp_dist/favicon.ico:webapp_dist/js/app.js.gz
    -Wall -Wextra
    ; disabled 
    ; -Werror

[env:TTGO_esp32_poe]
; http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1344&FId=t28:21:28
board = esp32dev
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=2
    -DHOYMILES_PIN_MOSI=15
    -DHOYMILES_PIN_SCLK=14
    -DHOYMILES_PIN_IRQ=34
    -DHOYMILES_PIN_CE=12
    -DHOYMILES_PIN_CS=4
    -DOPENDTU_ETHERNET
```

NetworkSettings.cpp (beginning at line 132)
Code is from TTGO example on github.
```
#ifdef OPENDTU_ETHERNET
    // Ergaenzugen fuer TTGO PoE Modul
    #define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT            // Version with not PSRAM
    // Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
    #define ETH_POWER_PIN   -1
    // Type of the Ethernet PHY (LAN8720 or TLK110)
    #define ETH_TYPE        ETH_PHY_LAN8720
    // I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
    #define ETH_ADDR        0
    // Pin# of the I²C clock signal for the Ethernet PHY
    #define ETH_MDC_PIN     23
    // Pin# of the I²C IO signal for the Ethernet PHY
    #define ETH_MDIO_PIN    18
    #define NRST            5

    pinMode(NRST, OUTPUT);
    digitalWrite(NRST, 0);
    delay(200);
    digitalWrite(NRST, 1);
    delay(200);
    digitalWrite(NRST, 0);
    delay(200);
    digitalWrite(NRST, 1);

    ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
    // ETH.begin();
#endif
```

Extension on the TTGO:
- 1
- 2
- 3  IRQ GPIO34
- 4
- 5
- 6
- 7
- 8 CE GPIO12
- 9 CS GPIO04
- 10 MOSI GPIO15
- 11 MISO GPIO02
- 12 SCLK GPIO14
- 13
- 14
- 15 3V3
- 16 GND

tbnobody, 2022-11-14:
Instead of adding source to the Networksettings.cpp you can also set the defines in the device environment in your platformio.ini

e.g. replace `ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);` with the original code, remove all the defines and just add in your platformio.ini:

```
-DETH_MDIO_PIN=18
-DETH_ADDR=0
```

etc... It would be pretty nice if the ethernet module would also work without toggling the ETH-PHY at startup. Then the board could be supported without any code modifications, just by adding a new environment

fromCologne, 2022-11-15:
Yes you are right. Better solution. My suggestion was only a dirty proof-of-concept ;-) because I struggled with wifi connection losses.

fromCologne, 2022-11-17:
The TTGO board is up and running since 5 days. With PoE.
I use a nRF board EBYTE E01-ML01DP5 because it´s shielded and has an external antenna. In germany delivered by the ´long river´.

tbnobody, 2022-11-18:
Oou I like the case. It's also available for the DIN rail. I just got such a board and testet it without any modifications in the `NetworkSettings.cpp` file. Just added the ports to the `platformio.ini`. Works perfectly!

fromCologne, 2022-11-19:
> Just added the ports to the platformio.ini. Works perfectly!
Great.

> Am I allowed to use your image as a example image for this board?
You are welcome.

tbnobody, 2022-11-19:
I would close this feature request as the changes are now applied in c0e6e13139fa6311ca668b9fbd75285887dabaab

closed 2022-11-19 by tbnobody