tbnobody/OpenDTU #337: Seriennummer wird nach Update nicht mehr akzeptiert

tobox, 2022-11-13:
### What happened?

Ich habe gerade meinen OpenDTU, die Ende September korrekt funktioniert hat, auf den aktuellen Github Stand geupdatet. Im Anschluss wurde der WR nicht mehr erkannt. Ich kann ihn auch nicht erneut einrichten, wenn ich die Seriennummer eingebe, wird gesagt, die Seriennummer wäre zu groß:

[image]

### To Reproduce Bug

Aktuelle Firmware einspielen, TSUN 1600 mit langer Seriennummer einrichten.

### Expected Behavior

Sollte funktionieren, so wie früher.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

bad9031

### Relevant log/trace output

```Shell
Siehe Screenshot
```

tbnobody, 2022-11-13:
Warum möchtest du die Seriennummer der DTU ändern? Hast du mehrere ESP Instanzen laufen und die Autogenerierung funktioniert nicht so das sich Seriennummern überschneiden?? 
Der Max Wert von 99999999999 wurde vor 19 Tagen auf 199999999999 geändert: 9c7dc681b1d91e58a80ccf268d56e3e335e04ee0

Frisch mal die Webseite neu auf (Strg+F5). Aber nochmal. An dieser Stelle trägt man NICHT die Seriennummer eines Inverters ein.

tobox, 2022-11-13:
Au man... User Error. Ich habe die Seriennummer des WR bei der DTU eintippen wollen... Mehrmals zu uralten Versionen gedowngraded, immer dasselbe Problem. 

Jetzt gehts, hatte nur Tomaten auf den Augen. Sorry.

closed 2022-11-13 by tobox