tbnobody/OpenDTU #350: API Daten auch per MQTT

DerFossiBaer, 2022-11-15:
### Is your feature request related to a problem? Please describe.

Ich hab über den String http://ip-vom-dtu/api/livedata/status die Totalwerte aller aller auf dem DTU angemeldeten Wechselrichter gefunden, so wie sie auch im Kopf der GUI angezeigt werden. Diese scheinen jedoch per MQTT versendet zu werden.

### Describe the solution you'd like

Die Daten habe ich mir jetzt per http-get geholt.
Schön wäre es, wenn die Daten direkt per MQTT mit auflaufen würden.

tbnobody, 2022-11-15:
Das steht bereits auf der "Wish List". Aber nun gibts auch ein Feature Request Issue das ich schließen kann wenn erledigt :)

fkroepfl, 2023-03-31:
@tbnobody die Frage des Neulings: wie sieht der aktuelle Plan dazu aus? 🤗

lanfire001, 2023-04-23:
Gibt es einen Zeithorizont wann das umgesetzt wird, oder muss man für dieses Feature auf die geforkte Version von Thro42 zurückgreifen?

tbnobody, 2023-04-24:
Implemented in 26cbc496a795454b6808913f4fe66f6c623a267c

closed 2023-04-24 by tbnobody