tbnobody/OpenDTU #363: HM-800 Hardware Part Number [Request]

da-Jimbo, 2022-11-20:
### Is your feature request related to a problem? Please describe.

Meldung in der Oberfläche: "Unknown model! Please report the "Hardware Part Number" and model"

### Describe the solution you'd like

ist ein HM-800
Hardware Part Number ist: 269561858
Hardware Version: 01.00

tbnobody, 2022-11-20:
Hallo, 
hast du zusätzliche Infos zu dem Inverter? Also ist er z.B. in irgend einer Weiße limitiert oder hat eine spezielle Herkunft?

da-Jimbo, 2022-11-20:
Servus,
Meines Wissens ist er nicht limitiert.
Hab das Set mit 2 Platten aber als 770Wp gekauft.
Die Leistung der beiden Platten ist 770Wp.
Gekauft bei Alpha-Solar

homeautomation2022, 2022-11-20:
Bitte mal die ganzen Infos aus Inverter Info hier posten, mal sehen, ob die FW auch eine andere ist.

Ob er 800W Nennleistung macht, kannst du herausfinden, indem du 100 Watt als Absolut non-persistent Limit sendest und dann mal 5min abwartest und anschließend nachsiehst, was bei Current Limit als Prozent vom Inverter zurückkam (100/800W = 12,5%).

da-Jimbo, 2022-11-20:
Model	Unknown model! Please report the "Hardware Part Number" and model (e.g. HM-350) as an issue
Detected max. Power	0 W
Bootloader Version	0.1.4
Firmware Version	1.0.8
Firmware Build Date	2020-07-10 12:58:00
Hardware Part Number	269561858
Hardware Version	01.00

Das mit den 800W testen probier ich die Tage mal aus.

da-Jimbo, 2022-11-20:
Der HM-800 muss aber deutlich über 600W machen, weil ich am 21.08.2022 die bisher höchste gemessene Leistung erreicht hab und zwar knapp 1340W.
Auf meinem Dach ist nur ein HM600 und der besagte HM800 installiert.
Ich mache den Test aber trotzdem und berichte dann

homeautomation2022, 2022-11-20:
OK, danke für die Info, wird dann ziemlich sicher ein normaler HM-800 sein, nur halt ein etwas äteres Modell, als die HM800 die bisher gemeldet wurden/bekannt sind(FW 1.0.10 von Ende 2021), wie sich an der FW ablesen lässt.
@tbnobody Der kann also als 800er eingepflegt werden.
@da-Jimbo Kann tagsüber ja nochmal über die 100W und den zurückgemeldeten Prozenten prüfen, ob er tatsächlich ungedrosselt ist, aber ich gehe stark von aus, auch nach dem Max-Messwert. Bisher gab es überhaupt erst nur einen 70% gedrosselten WR...

da-Jimbo, 2022-11-21:
Als Ergebnis der Drosselung zeigt er 13% an. Also passt das mit den 800Watt.