tbnobody/OpenDTU #378: Swagger UI for REST API documentation

kamuffe, 2022-11-23:
### Is your feature request related to a problem? Please describe.

Hi.

First of all thanks a lot of your effort and the team working so great to make this project possible.
I just got OpenDTU up and running with absolutely no microcontroller knowledge ;)

In the software world when in comes to REST API there is one candidate you can't get out of scope. 
Swagger UI.

Thanks a lot,
Kamuffe

### Describe the solution you'd like

It is possible to integrate Swagger UI for the REST APU documentation?

See https://swagger.io/tools/swagger-ui/

This way we could also export it and import open-api.json to Postman or Insomnia or even other software to talk to OpenDTU via REST without the need manually create eachand every request.

### Describe alternatives you've considered

N/A

### Additional context

N/A

CommanderRedYT, 2023-11-06:
I could think of writing the initial swagger docu, but I am not sure if I have time for maintaining it

CommanderRedYT, 2023-11-06:
API Endpoints from Code (2991af7):

```md
GET  /api/config/get
POST /api/config/delete
GET  /api/config/list
POST /api/config/upload

GET  /api/device/config
POST /api/device/config

GET  /api/devinfo/status

GET  /api/dtu/config
POST /api/dtu/config

GET  /api/eventlog/status

POST /api/firmware/update

GET  /api/gridprofile/status

GET  /api/inverter/list
POST /api/inverter/add
POST /api/inverter/edit
POST /api/inverter/del
POST /api/inverter/order

GET  /api/limit/status
POST /api/limit/config

POST /api/maintencance/reboot

GET  /api/mqtt/status
GET  /api/mqtt/config
POST /api/mqtt/config

GET  /api/network/status
GET  /api/network/config
POST /api/network/config

GET  /api/ntp/status
GET  /api/ntp/config
POST /api/ntp/config
GET  /api/ntp/time
POST /api/ntp/time

GET  /api/power/status
POST /api/power/config

GET  /api/prometheus/metrics

GET  /api/security/config
POST /api/security/config
GET  /api/security/authenticate

GET  /api/system/status
```

CommanderRedYT, 2023-11-06:
https://github.com/tbnobody/OpenDTU/blob/master/docs/Web-API.md is missing these:

```md
GET  /api/gridprofile/status

POST /api/inverter/order
```

stefan123t, 2024-07-14:
Updated the Swagger OpenAPI with more examples and including OpenDTU-OnBattery API endpoints / responses.

[opendtu_openapi.yaml](https://github.com/user-attachments/files/16228674/opendtu_openapi_20240714.txt)

CommanderRedYT, 2024-07-15:
Imo we should not include stuff in a openapi file in this repo if its not coded in this repo. onbattery is a fork. Also, it would make more sense to do a MR instead of uploading files

stefan123t, 2024-07-16:
I would like to gather some input on the REST API and the level of detail we should detail it in ?
Eg I haven’t added any possible responses yet, only request afaIk.

@schlimmchen Regarding OpenDTU-OnBattery this would be an option which has a more extensive set of API endpoints. But removing the superfluous stuff for OnBattery from the upstream OpenDTU should be easy IMHO.

@CommanderRedYT yes agreed but I am not easy at git using a fork/PR/gist yet and I think the Swagger OpenAPI is still too premature.
Can you check it out by downloading and renaming to opendtu_openapi.yaml or whatever you like?

CommanderRedYT, 2024-11-15:
@stefan123t I would propose the following:

- Not sure if in code or docs repo
- if in code repo, have it under `docs/`
- Mention it in the README.md (because not everyone reads the docs first)
- Mention it in the docs repo

Afaik swagger ui can just load yaml files.
But I would recommend the use of a GitHub action like this: https://github.com/marketplace/actions/swagger-ui-action

This one generates static html which we could then embed in the docs website.

stefan123t, 2024-11-15:
@CommanderRedYT I am not good at linking this github action into the docs repo.
I am happy that the PR #2410 worked out fine and hope that @tbnobody finds this useful.

@schlimmchen you may use the larger [opendtu-onbattery_swagger_openapi.yaml](https://github.com/user-attachments/files/16228674/opendtu_openapi_20240714.txt) which I added earlier.

CommanderRedYT, 2024-11-17:
@stefan123t I will try implementing it when I have time

tbnobody, 2025-02-07:
A little bit of api documentation was already added into the [docs repo](https://www.opendtu.solar/firmware/web_api/). tbd... going to close this here...

closed 2025-02-07 by tbnobody