tbnobody/OpenDTU #388: MQTT "limit_nonpersistent" werden nicht übernommen

technican1988, 2022-11-26:
### What happened?

Vielen Dank erstmal für die tolle Arbeit die hier geleistet wurde!
Beim Testen ist mir allerdings folgendes aufgefallen.

Wenn per MQTT das Powerlimit gesetzt wird werden die beiden Botschaften 
"[serial]/cmd/limit_nonpersistent_absolute" und
"[serial]/cmd/limit_nonpersistent_relative"
völlig ignoriert. 

Wenn man die Limits aber mit 
"[serial]/cmd/limit_persistent_absolute" oder 
"[serial]/cmd/limit_persistent_relative"
schreibt werden diese sofort übernommen.
Selbst die in der Doku erwähnten 4 minuten bei den absoluten Werten werden nicht benötigt.

Stören tut mich dieser Bug nicht wirklich, aber es ist mir eben aufgefallen.
Freuen tue ich mich sehr, dass die Übernahme der Werte in Echtzeit geschieht statt der angekündigten 4min - Hier könnte auch entsprechend die Doku angepasst werden.

DANKE :)

### To Reproduce Bug

Wert per MQTT über "[serial]/cmd/limit_nonpersistent_absolute" setzen.

### Expected Behavior

Wert wird nicht übernommen.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

fe16a92

tbnobody, 2022-11-27:
Ich  habe es gerade eben mehrfach getestet. Das setzen  über `/cmd/limit_nonpersistent_relative` und `/cmd/limit_nonpersistent_absolute` funktioniert problemlos.
Wenn du auf die serielle Konsole schaust solltest du auf folgendes sehen:
```
11:50:46.562 > Received MQTT message on topic: solar/116171603546/cmd/limit_nonpersistent_relative
11:50:46.562 > Limit Non-Persistent: 95 %
```

Bzgl. der Dokumentation... Dort steht:
`The updated value will show up in the web GUI and limit_relative topic after around 4 minutes.`
Das bedeutet, das der aktualisierte Wert nach 4 min in der Web GUI angezeigt wird. Im Inverter angewendet wird er nahezu sofort. Das mit den 4 Minuten gilt aber auch nur für Inverter welche über ihre Hardware ID nicht identifiziert werden können. Ich werde die Doku hier anpassen.

openKimchi, 2022-12-11:
Ist komisch, bei mir klappt das mit den nonpersistent Werten auch nicht. Persistent klappt einwandfrei.

tbnobody, 2022-12-12:
Was ist deine Base topic?
An welche Topic published du deine Werte?
Was sagt die Serielle Konsole?

openKimchi, 2022-12-12:
Base topic ist solar/
ich publishe meine Werte an:
solar/serial/cmd/limit_persistent_relative  (funktioniert)
oder an:
solar/114163906562/cmd/limit_nonpersistent_relative  (funktioniert nicht)

Wie rufe ich die serielle Konsole für MQTT auf?

tbnobody, 2022-12-13:
Den ESP via USB mit dem Rechner verbinden, putty Laden, auf seriellen Modus umschalten, com Port eintragen, baudrate auf 115200 setzen
Wenn du es selbst mit VSCode compiliert hast, geht es noch einfacher. Dort gibt es in der Statusleistung einfach ein Icon welches die Serielle Konsole öffnet.

openKimchi, 2022-12-13:
OK, danke! Werde ich mal versuchen.
Ich habe nur irgendwie den Eindruck, dass es irgendwas mit dem Inverter selbst zu tun haben könnte.
Es ist nämlich so, dass, wenn ich den nonpersistent Wert aus der Web Gui aus setze, der zwar scheinbar angenommen wird, an der Leistung des Inverters ändert sich aber nichts. Im Gegensatz zu den persistenten Werten...

tbnobody, 2022-12-14:
Dann sende den Wert bitte nicht als retained. Ich habe das eingebaut weil wenn beide Topic (persistent / nonpersistent) retained published werden ist es Zufall welcher Wert zuerst kommt und das komplette Ergebnis ist undefiniert.

openKimchi, 2022-12-14:
Wie setze ich denn in ioBroker den Wert als nicht retained? Oder reicht es, in der Web GUI das Retained Flag nicht zu setzen?

tbnobody, 2022-12-14:
das kann ich so leider nicht beantworten, da ich kein ioBroker verwende. Aber das mit der GUI klingt erstmal gut.

openKimchi, 2022-12-14:
ich meinte die GUI von openDTU. Aber das funktioniert schonmal nicht...

tbnobody, 2022-12-14:
ne, das Setting in der OpenDTU beschreibt ja nur wie OpenDTU selbst die Werte Published... Du musst das schon im ioBroker irgendwo einstellen

openKimchi, 2022-12-14:
OK, danke! Aber bin ich ja schon ein Stück weiter.

openKimchi, 2022-12-15:
So, um das Ganze zum Abschluss zu bringen: Im MQTT Adapter für den ioBroker kann man eine Option anhaken: "Nachrichten ohne "retain"-Flag senden".
Hakt man das an, werden die Werte übernommen.

tbnobody, 2022-12-15:
Danke! Dann würde ich das Issue hier schließen. Eine Anpassung der Doku, bei der dieses Verhalten erwähnt wird, werde ich demnächst pushen.

Jean-GO, 2023-05-25:
Hallo, ich habe den Hoymiles HMT-2250-6T und bei mir funktioniert das setzten des Limits über MQTT auch nicht.
Ich habe beide Varianten getestet, es hat sich nichts geändert.. (iobroker mit MQTT)
Hat schon einer eine Lösung?
Oder gibt es ein 
http://...IP.../limit_nonpersistent_relative
Befehl