tbnobody/OpenDTU #391: [Request] MI-1500 Hardware Part Number

andrerdrummond, 2022-11-27:
### Is your feature request related to a problem? Please describe.

Inverter Info

Model	Unknown model! Please report the "Hardware Part Number" and model (e.g. HM-350) as an issue [here](https://github.com/tbnobody/OpenDTU/issues).
Detected max. Power	0 W
Bootloader Version	0.1.0
Firmware Version	1.0.12
Firmware Build Date	2020-06-24 19:00:00
Hardware Part Number	268578817
Hardware Version	06.00

[image]

### Describe the solution you'd like

Include MI-1500 at device info list

andrerdrummond, 2022-11-27:
Looks like this issue: https://github.com/tbnobody/OpenDTU/issues/363

homeautomation2022, 2022-11-27:
@tbnobody @andrerdrummond This is one of the very few 1062xxxxxxxx MI-1500s, these are the only ones the MI's use the Gen3 protocol, that should be included, you're right. Thanks for reporting :-)

tbnobody, 2022-11-28:
Thank you for the hint! I added the hardware ID in: 3c8bd5657699990299a55c39b6fb5592353094c6

closed 2022-11-28 by tbnobody