tbnobody/OpenDTU #397: [Request] Nachtabschaltung Deep Sleep

rockermeier, 2022-11-28:
### Is your feature request related to a problem? Please describe.

Wäre es möglich, eine Option zu implementieren, die den ESP über Nacht in einen Deep Sleep versetzt? Zwar wäre die OpenDTU nicht mehr ansprechbar, bei allen, die die Daten über MQTT loggen, ist das aber höchstwahrscheinlich sowieso irrelevant. Stattdessen könnte nicht gerade wenig Strom eingespart werden, nachdem mein WT31-ETH01 schon über 2W braucht. 
In einem Pull request  (siehe #351 ) wurde schon eine Lösung zur Berechnung des Sonnenaufgangs und -untergangs zur Deaktivierung des NRF-Moduls vorgeschlagen. Das alleine würde aber sicherlich keine signifikanten Einsparungen hervorbringen.

### Describe the solution you'd like

Option, mit der der ESP32 über Nacht in den Deep Sleep versetzt wird, um den unnötigen Stromverbrauch zu senken.

Solarteur, 2022-11-28:
Das Thema wurde hier 
https://github.com/tbnobody/OpenDTU/discussions/265
schom einmal diskutiert.

 '>2W erscheinen mir viel. Welche Leistungsaufnahme hat dein Netzteil im Leerlauf? Bei mir läuft die komplette OpenDTU-Einheit (ESP32 Dev. Board und nRF24L01+) mit < 1W.

rockermeier, 2022-11-29:
Dass das Thema schon einmal diskutiert wurde, habe ich nicht bemerkt, zeigt aber, dass mehrere das Anliegen haben. Ich verstehe nicht ganz, warum eine Nachtabschaltung ein Problem darstellen sollte. 
Zum Stromverbrauch: Mein WT32 braucht mehr, hat aber Ethernet. Woran das genau liegt, weiß ich nicht, evtl. hat das Board auch einen Fehler. 
Aber auch bei nur 1W Verbrauch könnte man jährlich immerhin gut 4kW einsparen. Besser als nichts, v.a. wenn man die Summe der vielen Nutzer betrachtet.

scorpi11, 2023-01-17:
V.a. wenn oft nur ein kleines Balkonkraftwerk dranhängt würde ich auch versuchen, den verursachten Zusatzverbrauch zu minimieren. Ich habe hier einen zweiten ESP32 zum Spielen und werde mal versuchen, darauf einen zeitgesteuerten Deep Sleep-Betrieb hinzubekommen.

scorpi11, 2023-01-17:
Mit einem ESP32 NodeMCU geht der Timer-Wakeup hier zuverlässig. Der Stromverbrauch geht dann auf dem Messgerät von 0,4W auf 0,0W (nacktes Gerät ohne aktives WLAN und ohne NRF24L01+-Funkmodul). Ich werde mal versuchen, das mit dem Code der Berechnung von Sonnenauf- und Untergang zu verheiraten.

scorpi11, 2023-01-19:
Sodele, jetzt habe ich alle Puzzleteile beisammen.

Zunächst ein mal eine Funktion, um die Minuten bis zum nächsten Sonnenaufgang zu berechnen, basierend auf dem eingangs erwähnten Pull-Request.

Dann habe ich mir eine modifizierte OpenDTU-Firmware gebaut, die immer 10 Sekunden nach dem Booten für 20 Sekunden in den Sleep-Modus geht, und außerdem mit der blauen LED anzeigt, ob das Funkmodul gefunden wurde (um mögliche Probleme nach dem Aufwachen auszuschließen). Das funktionierte auch für ein paar Minuten zuverlässig. Dabei wurden im Betrieb 1,3 W und im Sleep-Modus 0,0W auf dem Messgerät angezeigt.

Da könnten sich also durchaus ein paar kWh pro Jahr sparen lassen, auf jedem OpenDTU-Gerät weltweit! 😎

Als nächstes werde ich voll funktionstüchtige Firmware bauen, die sich nach Sonnenuntergang schlafen legt und zum Sonnenaufgang wieder aufwacht.

LustigePerson, 2023-01-20:
Finde ich eine super Initiative von dir. Der Gedanke kam mir auch nachdem ich diese Woche mit mienem Balkonkraftwerk eingestiegen bin.
Perfekt wäre eine Gewisse Karenz (konfigurierbar), z.B. Boot X minuten vor Aufgang und shutdown X Minuten nach Untergang.
Wenn du irgendwo nen Fork hast teste und schaue ich gerne mit!

scorpi11, 2023-01-20:
Das ist in dem Fork, den ich als Grundlage benutze, schon genau so drin (siehe PR #351). Wenn mein OpenDTU-Gerät morgen früh wieder wie gewünscht aufwacht, werde ich meine Änderungen veröffentlichen und gebe hier Bescheid.

scorpi11, 2023-01-21:
Der Wakeup heute Morgen ging wie gewünscht. Hier der Code dafür: https://github.com/scorpi11/OpenDTU/tree/sleep-at-night

muibusan, 2023-02-04:
Wird hoffentlich auch Teil des Projekts von tbnobody. :) Sonst haben wir am Schluß 100 Forks.

testos77, 2023-02-17:
> Der Wakeup heute Morgen ging wie gewünscht. Hier der Code dafür: https://github.com/scorpi11/OpenDTU/tree/sleep-at-night

can confirm, compiled your fork -  everything worked as expected. Thx
Hopefully this option will be integrated in the main code.

FilipeAmadeuO, 2023-02-19:
Excelent. Can it be integrated in main code ?

LarsRaw, 2023-02-20:
Ich habe den Forck auch compiliert und DeepSleep funktioniert einwandfrei. Es macht Sinn, denn bei mir im Garten (Insel) zählt im Winter jede Wattstunde :) 

Ich hoffe wirklich es wird in den Main Code integriert und nicht nur NRF inaktiv schalten wie in der Roadmap beschrieben... Können wir noch was machen das @tbnobody den Thread liest? Es wäre ja nur ein Schalter mehr. Einen für DeepSleep und einen nur für NRF pollen inaktiv. Dann kan jeder machen was er möchte :) Mir ist DeepSleep wichtiger das WLAN dan auch aus ist :)

chris-wein, 2023-02-26:
Gibt es den Code mit der Nachabschaltung zum Aufspielen als Over-The-Air-Update? Ich bin leider kein Programmierer und habe nicht herausfinden können, wie ich aus dem reinen Softwarecode eine Aufspielbare-Datei erstelle..

Freu mich auf Hilfe!

LarsRaw, 2023-02-26:
Den aktuellen Code mit DeepSleep gibts hier zum runterladen. Funktioniert auch per OTA:

https://github.com/scorpi11/OpenDTU/actions/runs/4145669841

chris-wein, 2023-02-26:
Super danke! Da sind vier .bin-Dateien. Müssen die alle installiert werden und wenn ja, in welcher Reihenfolge?

Solarteur, 2023-02-27:
Das Vorgehen ist hier beschrieben: https://github.com/tbnobody/OpenDTU#flash-with-espressif-flash-download-tool-windows

Wenn OpenDTU schon auf dem ESP32 installiert ist, reicht wahrscheinlich das Installieren der "opendtu-generic.bin" über "Settings" --> Firmware upgrade".

Aber: https://github.com/scorpi11/OpenDTU/actions/runs/4145669841 ist ein Fork von "[tbnobody ](https://github.com/tbnobody)/
[OpenDTU](https://github.com/tbnobody/OpenDTU)", d. h. die Nachtabschaltung ist realisiert, nicht aber die Änderungen, die zwischenzeitlich in "tbnobody/OpenDTU" vorgenommen wurden.

chris-wein, 2023-02-27:
Cool, super! Echt genial! Tatsächlich hat es allein mit dem Firmwareupgrade funktioniert, allerdings muss man einen Haken setzen bei "Sunrise". War mir nicht ganz klar, aber jetzt läufts einwandfrei.

Weiß jemand, ob noch relevante Features zwischen dem Fork und der Hauptversion hinzugekommen sind?

joschua234, 2023-03-08:
Habe das Upgrade mit Nachtabschaltung vor einigen Tagen aufgespielt und muss sagen es funktioniert wie erwartet und ist einfach nur super.
Wäre echt toll, wenn es euer Code in den Standard Code schaffen würde.
Vielen Dank für eure Arbeit :)

fow0ryl, 2023-03-16:
Hallo, habe diese [Image](https://github.com/scorpi11/OpenDTU/actions/runs/4145669841) vor 2 Tagen eingespielt.  
Gestern Nachmittag musste ich wegen notwendiger Montagearbeiten den Wechselrichter vom Netz trennen. Als ich ihn wieder ans Netz genommen habe, wurde es bereits relativ dunkel, so das kein Strom mehr produziert wurde. OpenDTU war aber per http zumindest bis Mitternacht erreichbar.
Heute Morgen 09:55 hat der ESP32 aber keine WLAN Verbindung mehr. Auch auf meinem Router erscheint er nicht mehr in der Liste, die eine DHCP Anfrage gestellt haben. WLAN scheint also komplett tot zu sein :(.

LarsRaw, 2023-03-20:
@fow0ryl Ich glaube du hast dich bei den Koordinaten vermacht, so das sich der Deepsleep verschoben hat.

scorpi11, 2023-03-20:
In der Konsole sieht man alle paar Sekunden die berechnete Zeit für Sonnenauf- und Untergang, sowie die verbleibende Zeit bis zum nächsten Sonnenaufgang. Diese Zeiten sollten passen, ansonsten stimmt etwas mit den Einstellungen nicht.

LarsRaw, 2023-03-22:
@scorpi11 Da sich bei beim Standart Code sowohl der Bootloader und die Sunsetberechnung nativ integriert worden ist, wäre es möglich für dich einen neuen Build zu machen. Ein hinher switchen zwischen deinen Build und den Standart ist per OTA ist wegen den geänderten Build nicht mehr möglich... Wäre schön wenn du einen akruellen Build machen könntest, ist ja nur den einen Schlater für DeepSleep integrieren :) Vielen Dank

fow0ryl, 2023-03-22:
Mein OpenDTU war zwar im WLAN angemeldet, hat aber per DHCP über Tage keine IP angefordert. Demzufolge war er  per http auch nicht mehr erreichbar war. Ich  habe letztlich die Spannungsversorung zum ESP32 gekappt und nach einer Weile wieder hergerstellt.
Ich habe dann die Einstellungen geprüft. Die Koordinaten passen und auch die auf der Konsole ausgegebenen Zeiten sind ok.
`
11:58:13.533 > Sunrise: 6:16
11:58:13.533 > Sunset: 18:33 
`

Das Gute: Seit dem letzten Hard-Reset sind keine Störungen mehr aufgetreten...

scorpi11, 2023-03-23:
@LarsRaw Danke für die Info, dann werde ich mal mein Git-Repository anpassen.

dj9717, 2023-03-27:
Hallo,
hab mir das GIT Repo https://github.com/scorpi11/OpenDTU.git geholt, kompiliert und auf den ESP hochgeladen.
Kann/muss man die DeepSleep Funktion irgendwo aktivieren ? Ich finde da nichts ?

Grossen Dank für den Fork.
Ich hatte vor Monaten das DeepSleep Feature angefragt. Leider erfolglos.

LarsRaw, 2023-03-27:
aktivieren unter Einstellungen -> NTP Einstellungen dei RICHTIGEN! koordinaten eingeben und dann den Schlater "Sunrise" aktivieren :)

LarsRaw, 2023-03-27:
[image]

dj9717, 2023-03-27:
Der Abschnitt "Sunrise Sunset Einstellungen" fehlt bei mir ?!

dj9717, 2023-03-27:
Oh ok.
Der master branch enthält die Funktion vermutlich nicht. Ich versuche es mal mit dem sunset+sleep Branch.

dj9717, 2023-03-28:
**Super Feature** **tausend Dank** nochmal für die Umnsetzung.
Funktioniert wunderbar.
Ich hoffe sehr das dies nun doch noch Einzug in den Hauptzweig hält.
Bisher wurde die Notwendigkeit zur Energieeffizienz dort leider sehr gering bewertet.

marco050969, 2023-04-16:
Hi,
ich hatte die Nachtabschaltung auch getestet, hat prima funktioniert, nur leider konnte ich mit diesem Build nicht mehr die einzelnen Panelwerte und die Temperatur vom Wechselrichter auslesen über iobroker, wieder zurück zum gerade aktuellem funktionierte wieder alles.
Neustarten hat leider auch nichts geholfen.

dj9717, 2023-04-16:
> Hi, ich hatte die Nachtabschaltung auch getestet, hat prima funktioniert, nur leider konnte ich mit diesem Build nicht mehr die einzelnen Panelwerte und die Temperatur vom Wechselrichter auslesen [...]

Ist halt leider so, das sich @tbnobody schon vor @scorpi11 's Fork, gegen dieses zeitgemäße und nützliche Feature des ESP32 entschieden hat. Seine Begründung findest Du hier https://github.com/tbnobody/OpenDTU/discussions/265.

Jamie-Jameson, 2023-06-01:
wäre es möglich, den aktuellen Stand vom tbnobody:master zu mergen? Der sleep-at-night Branch sei 350 commits hinterher...

Ansonsten würde ich mir auch wünschen, dass ein PR ins Hauptprojekt möglich wäre.

Jamie-Jameson, 2023-06-01:
Ja, das verstehe ich, klaro. Ich habe die 350 Commits grob überflogen. Vieles dreht sich ums Display (brauch ich nicht), aber es gab auch einiges zu Config und MQTT, glaube ich.

Andy-Op, 2023-06-09:
Es gibt ein neues Build von @tbnobody. Könntest du das mit Nachtabschaltung bereitstellen? 

> @Jamie-Jameson
> 
> Mach doch vielleicht noch mal einen neuen Request auf. Ich werde das supporten. Vllt können wir mit Argumenten überzeigen und @tbnobody überzeugen. Man könnte ja ein zusätzlich Bestätigung vorm einschalten abfragen, damit nicht "unwissende" User das aus versehen aktivieren und die DTU nachts nichts mehr erreichbar ist...

testos77, 2023-07-02:
> Es gibt ein neues Build von @tbnobody. Könntest du das mit Nachtabschaltung bereitstellen?
> 
Have a look at https://github.com/OFreddy/OpenDTU 
atm only 70 commits behind and includes the new build. Settings in NTP section:
[image]

Andy-Op, 2023-07-24:
I'll use v23.7.22. There is no switch in NTP section. Where do i find the needed binary-file?
[image]

chriszuercher, 2024-03-16:
@tbnobody It would be realy nice if this could be integrated into the main repository. Even if only e few users will use it (what I doubt) it will help them to save some energy. Which is basically one of the ideas of this project I guess? Produce green energy and dont't waste it?

If there are no points against it and if it works, why not just doing it?

stefan123t, 2024-07-10:
There are two things in this Feature Request:

1. Deep Sleep function while the Inverters are unreachable to conserve power used by the OpenDTU
2. Specific Offsets to switch into Deep Sleep after dusk/before dawn.

I have checked both the code suggested for deep sleep by @scorpi11 and @OFreddy and the current implementation of sunset types (below).

1. The actual Deep Sleep code by Tino Mettler is in https://github.com/tbnobody/OpenDTU/commit/d06b307d9d9ee338c3ec8aa725c3a07052b14336 and there is also a Wake on LAN (WOL) feature too implemented by Tino Mettler in @scorpi11's fork: https://github.com/tbnobody/OpenDTU/compare/master...scorpi11:OpenDTU:sleep-at-night

Is the Wake-On-LAN code sufficiently stable to wake the OpenDTU from Deep Sleep, in case we want to make some changes to the OpenDTU configuration, e.g. disable Deep Sleep again ?

2. The current Sunset type allow to set an offset after dusk / before dawn. However the effect may be a bit unclear as the resulting Sunrise / Sunset times are shown on the Info > NTP page after saving the Sunset type on the Settings > NTP page.
If possible the Sunrise / Sunset time could be shown on the Settings > NTP page already ?

Sunset type: Standard dawn (90.8°)
Local Time	Wednesday, July 10 2024 08:38:28
Sunrise	Wednesday, July 10 2024 05:30:00
Sunset	Wednesday, July 10 2024 21:20:00

Sunset type: Nautical dawn (102°)
Local Time	Wednesday, July 10 2024 08:39:09
Sunrise	Wednesday, July 10 2024 03:53:00
Sunset	Wednesday, July 10 2024 22:56:00

Sunset type: Civil dawn (96°)
Local Time	Wednesday, July 10 2024 08:37:07
Sunrise	Wednesday, July 10 2024 04:49:00
Sunset	Wednesday, July 10 2024 22:01:00

Sunset type: Astronomical dawn (108°)
Local Time	Wednesday, July 10 2024 08:39:47
Sunrise	Wednesday, July 10 2024 02:28:00
Sunset	Thursday, July 11 2024 00:20:00

I assume setting the Civil dawn (96°) or even Nautical / Astronomical dawn should allow for enough offset.

Or do we need a distinction of the offset for deepsleep after sunset (dusk) and before sunrise (dawn) because people want to keep the OpenDTU available until e.g. midnight and start it up again shortly before dawn ?

scorpi11, 2024-07-10:
The Wake On Lan code slipped in by accident, it is unrelated to the deep sleep function. It also has nothing to do with waking up the OpenDTU device via WOL. Sorry for this.

kaeferfreund, 2024-09-20:
Any update on this? Anything where help is needed?

stefan123t, 2024-09-21:
@kaeferfreund @scorpi11 before you invest any time and effort into this feature you should get the management buy-in. I do not see this PR going anywhere because the main developer and maintainer does not want to have this feature because it may introduce additional issues while it does not yield many benefits. 
See his comments here https://github.com/tbnobody/OpenDTU/discussions/265#discussioncomment-4115751

Currently there are also only 9 upvotes on the first entry in this issue / Feature Request.