tbnobody/OpenDTU #400: WLAN Anmeldung an ESP32 AP funktioniert nicht mehr

FrodoVDR, 2022-11-29:
### What happened?

Ich kann mich nach einer Neuinstallation auf einem noch nicht für OpenDTU benutzten ESP nicht an dessen AP verbinden.
Egal ob ich mein iPhone, PC oder Android Tablet verwende bekomme ich immer einen Authorization Fehler.

### To Reproduce Bug

Übertragen und gebaut habe ich OpenDTU mit Visual Studio Code unter Windows 11.

### Expected Behavior

Die Anmeldung funktioniert wieder oder ein Workaround um den Source Code direkt mit meinem WLAN zu verbinden.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

40203f7

### Relevant log/trace output

```Shell
--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
12:25:56.500 > 
12:25:56.500 > Starting OpenDTU
12:25:56.500 > Initialize FS... done
12:25:56.500 > Reading configuration... done
12:25:56.500 > Initialize Network... done
12:25:56.500 > Setting Hostname... Initialize NTP... done
12:25:56.500 > Initialize MqTT... done
12:25:56.500 > Initialize WebApi... done
12:25:56.500 > Check for default DTU serial... done
12:25:56.500 > Initialize Hoymiles interface... Connection error!!
12:25:56.500 >   Setting radio PA level... 
12:25:56.500 >   Setting DTU serial... 
12:25:56.500 >   Setting poll interval... 
12:25:56.500 > done
12:25:56.500 > Switch to WiFi mode
12:25:56.500 > Setting Hostname... done
12:26:10.597 > Disable search for AP... done
12:36:12.189 > Enable search for AP... done
12:36:12.194 > Setting Hostname... done
12:36:28.204 > Disable search for AP... done
```

tbnobody, 2022-11-29:
War der ESP neu? Falls nein, hast du ihn vorher komplett gelöscht?

FrodoVDR, 2022-11-29:
Ich muss das neu erleutern, nachdem ich den ESP32 per "ERASE" geöscht habe kann ich mich nun an den ESP32 über dessen AP auf IP 192.168.4.1 verbinden. Trage ich nun meine SSID und Passphrase ein verbindet sich OpenDTU damit, allerdings bleibt weiterhin der AP aktiv. Scheinbar deaktiviert der sich gar nicht mehr. Denn man kann sich weiterhin mit IP 192.168.4.1 verbinden aber nicht mit der über Networkinfo oder in der Consolen Ausgabe angezeigten. Auch pingen kann man diese IP Adresse nicht.

FrodoVDR, 2022-11-29:
Zu Deiner Frage er war neu.

tbnobody, 2022-11-29:
Der Access Point bleibt 3 min, nachdem sich der ESP mit einem WLAN verbunden hat, offen.

> Auch pingen kann man diese IP Adresse nicht.

Du bist zu diesem Zeitpunkt allerdings schon im korrekten Netz oder?

FrodoVDR, 2022-11-29:
Ja, wenn ich NTP Sync mache klappt das, im Info bekomme ich auch alle meine per DHCP empfangene Netzwerk Parameter angezeigt. Auch der Zugriff auf den Wechselrichter funktioniert, dann. 

Nur der Zugriff und Ping auf die eigen IP Adresse ist Glückssache.

Ich kam jetzt auch mal auf das richtige lokale Netzwerk Interface, aber nach einspielen des Backups ist die Verbindung wieder kaputt. Das OpenDTU AP Passwort ist dann auch geändert auf meines von meiner SSID.

Probiert habe ich das mit zwei ESP32 bei beiden passiert das selbe.

tbnobody, 2022-11-29:
Welches Backup versuchst du einzuspielen? Eine .json Datei oder eine .bin Datei? 
Wenn du die .json Datei in einem Texteditor öffnest, stehen dort die korrekten Wifi Zugangsdaten?

tbnobody, 2022-11-29:
> Das OpenDTU AP Passwort ist dann auch geändert auf meines von meiner SSID.

Den Satz verstehe ich nicht ganz. Zu Beginn ist das OpenDTU AP Passwort ja "openDTU42". Genauso wie das Login Passwort im Webinterface. (Beide Passwörter sind immer identisch, also das Eine entspricht immer dem Anderen).

Dann gibst du im "Network Settings" Dialog die SSID sowie das Passwort von deinem Router ein. OpenDTU verbindet sich dann mit dem Router und bekommt eine IP aus dessen Subnetz. 
Anschließend verbindest du dich mit der IP die der ESP vom Router bekommen hat mit dem Webinterface und stellst ein Backup wieder her. 
Daraufhin verbindet sich der ESP nicht mehr mit der Router SSID und das Passwort für das WLAN das der ESP selbst aufmacht, sowie das Passwort für das Webinterface entspricht dem Passwort was du vorher im Netzwerk Dialog eingegeben hast?

FrodoVDR, 2022-11-29:
Ja ich habe das admin Paswort geändert, dass das vom admin Passwort abhängt hatte ich bisher nicht realisiert.
Das erklärt zumindest warum der AP Modus ein anderes Passwort bekommen hat.

Ich habe es nun hinbekommen das der neu aufgesetzte ESP32 funktioniert, auch wenn ich nicht verstehe was da los war. 

Die Einstellungen habe ich von Hand vorgenommen, beim Vergleich der alten und der neuen json Backupdatei habe ich nur gefunden das ich den Level noch nicht von Low auf Minimum gesetzt hatte.

Vorzorglich habe ich nun noch die IP Konfiguration manuell eingetragen.

Ich werde mir bei Gelegenheit eine zweite OpenDTU aufsetzen ich habe noch zwei ESP32 und zwei Funkmodule.

Das komische an der ganzen Sache ist das bis gestern alles reibungslos funktionierte (der letzte Update hatte ich in der letzten Woche eingespielt), heute war auf einmal die OpenDTU nicht mehr erreichbar und liefert auch keine Daten mehr über MQTT. Das war der Punkt alles neu aufzusetzen, erst mit der verhanden alten OpenDTU und als das nicht funktionierte mit einer neuen. Das dann beides nicht gleich funktionierte bewog mich dazu hier den Bug zu reporten. 

Irgend etwas ist zumindest nicht in Ordnung, als ich das erstmals aufgesetzt habe mitte Oktober funktionierte alles beim ersten Mal. Entweder blockiert mein Router hier etwas oder in der OpenDTU hat sich am Netzwerkhändlig etwas geändert.

tbnobody, 2022-11-29:
Die letzten Änderungen am Netzwerk Handling gab es am 29. Oktober: https://github.com/tbnobody/OpenDTU/commits/master/src/NetworkSettings.cpp

Dort habe ich aber nur etwas am String handling geändert und nichts an der Logik. Alle anderen Änderungen siehst du im Link. Es wurde schon sehr lange nichts mehr am Netzwerk Handling geändert. 

Als Hinweis, direkt unter dem Eingabefeld für das Admin Passwort sollte folgender Text stehen:
`The administrator password is used to access this web interface (user 'admin'), but also to connect to the device when in AP mode.`

Wenn dieser bei dir nicht vorhanden ist, refreshe doch das Webinterface mal mit Strg+F5

FrodoVDR, 2022-11-29:
Der Text ist vorhanden, den hatte ich übersehen.

Jetzt funktioniert auch der zweite ESP32 wieder, vermutlich hat da etwas in meinem WLAN geklemmt.
Auch die Updates auf den letzten Commit funktionierte ohne Probleme, somit schliesse ich diesen "vermeintlichen" Fehler.

closed 2022-11-29 by FrodoVDR