tbnobody/OpenDTU #402: Hoymiles HM1500 not supportet

cvggithub, 2022-12-01:
### What happened?

Hello, 

I loaded the latest build but my HM1500 is not supportet . i just see two strings and no values with my device.
radio is working 

### To Reproduce Bug

I don´t know what I can say here if you have not my HM1500 Device 

### Expected Behavior

I would like to see 1 Pahse information and 4 Strings

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

e8c517f

tbnobody, 2022-12-01:
What is the serial number of your inverter?

cvggithub, 2022-12-01:
Hallo, offensichtlich war ICH zu blöd, die richtige Seriennummer einzutragen. Mein Fehler,  aber höchsten Respekt für diese megaschnelle Antwort. 3 Daumen hoch 
LG Christian

closed 2022-12-01 by cvggithub