tbnobody/OpenDTU #404: WiFi - Verbindungsprobleme ESP32 zur Fritz!Box

hoffmakl1961, 2022-12-02:
### What happened?

Ich habe WiFi - Verbindungprobleme, wenn sich meine ESP32 mit dem WLAN-AP meiner Fritz!Box 7490 (FRITZ!OS: 07.29 - Version aktuell) verbinden. 

Das Problem ist, dass beim ESP32 nur jeder 2. WLAN Verbindungsversuch erfolgreich ist.  

### To Reproduce Bug

Die Ursache ist von ullixesp » Fri Apr 30, 2021 9:10 am beschrieben: https://www.esp32.com/viewtopic.php?t=16622. 

Die meisten modernen Fritz!Boxen erfordern - wenn auch nicht ausschließlich - einen Double-Hitter-Ansatz, um eine WiFi-Verbindung herzustellen. Das heißt, man macht einen ersten Versuch, der fast immer fehlschlägt, dann versucht man es erneut, was "fast" immer gelingt.  

### Expected Behavior

Die Probelmlösung ist ein Software Reset: 
Nach einer Zeitüberschreitung in der Verbindungsroutine  (z.B. 8 erfolglose Verbindungsversuche) wird ein "ESP.restart();"  ausgeführt.

Das sieht bei mir so aus. 

  while(WiFi.status() != WL_CONNECTED){
    delay(500); 
    Serial.printf(".");
    ++CounterWLAN;
    if(CounterWLAN > 8) {  
      Serial.println();
      ESP.restart(); }
  }
  
Könntet ihr das mit einbauen, um diesem bug zu umgehen? 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

Git SHA: dec333f :: 0.5.41

### Relevant log/trace output

```Shell
nicht vorhanden
```

tbnobody, 2022-12-02:
1. Ein Git Hash dec333f oder Version 0.5.41 existiert hier nicht. Meinst du ggf. das Ahoy Projekt? Dann ist das hier aber falsch.
2. Wenn keine Verbindung aufgebaut werden kann findet bei OpenDTU sowieso schon zyklisch ein neuer Versuch statt. Warum sollte man zwischen den Versuchen neu booten wollen?

hoffmakl1961, 2022-12-02:
Ja, das Ahoy Projekt. 
Die WLAN Verbindung wird unter den o.g. Bedingungen nicht aufgebaut, auch wenn es zyklisch immer wieder versucht wird. 

Erst nach einem Reboot. Das ganze ist reproduzierbar.

tbnobody, 2022-12-02:
Dann würde ich dich bitten den Fehler auch beim Ahoy Projekt zu melden und nicht hier.

hoffmakl1961, 2022-12-02:
Ich melde es im Ahoy-Projekt. 
Soll ich den Issue hier closen?

closed 2022-12-02 by tbnobody

hoffmakl1961, 2022-12-02:
Eröffnet im Ahoy Projekt: https://github.com/lumapu/ahoy/issues/471