tbnobody/OpenDTU #412: HMS -series support

Mart-Muc, 2022-12-07:
### Is your feature request related to a problem? Please describe.

Hab mich grad gewundert, warum der HM600 plötzlich so im Preis verfallen ist..

siehe da- es gibt ne neue HMS-Serie

Wie schaut da das Protokoll aus?
Ist das kompatibel?

Danke

PS: wann kam denn die neue Serie raus? finde nichts über den Modelwechsel/Update bei google...strange

### Describe the solution you'd like

Hat jemand schon einen HMS und arbeitet an der Integration?

Solarteur, 2022-12-07:
Sieht z. Z. nicht gut aus, vgl. https://github.com/tbnobody/OpenDTU/issues/46

homeautomation2022, 2022-12-09:
So schlecht siehts garnicht aus, siehe im Discord, Protokoll scheint sehr ähnlich zu HM geblieben zu sein, nur auf anderer Frequenz. Obs dieses Jahr noch funktionell wird, mal sehen...

mbay0r, 2022-12-26:
Habe auch einen HMS-2000 gekauft und warte gespannt. Nehme jetzt erst mal eine DTU Pro S

DevYves89, 2023-04-05:
Any updates on this one ?

tbnobody, 2023-04-05:
Still work in progress. But it is really work going on. It's nothing which is on a todo list to implement later. I am currently actively working on it.

DevYves89, 2023-04-05:
Thank you really for your time and effort. I really appreciate it.
Iam going to implement 4+ of the HMS series next month and would love to use my current opendtu setup for it. If i can help in testing ie. new images later on, iam willing to do so. Also donate something for your coffee. Thank you.

Floris039, 2023-04-19:
I saw that I was able to implement the HM-2000 4T with the EBYTE E49-900M20S. Somebody any idea where to buy that? Because the NRF240L01+ is easily found but the EBYTE not. I live in the Netherlands.

tbnobody, 2023-04-23:
I am going to close this Feature Request as the HMS series is supported now.

closed 2023-04-23 by tbnobody

sarmbruster, 2023-04-29:
> For example here: https://www.aliexpress.com/item/1005003290274214.html?spm=a2g0o.productlist.main.47.480030f273bJsY&algo_pvid=aca58083-a120-45c0-930a-86a1b4863a81&algo_exp_id=aca58083-a120-45c0-930a-86a1b4863a81-23&pdp_npi=3%40dis%21EUR%212.63%212.6%21%21%21%21%21%402100bfe316816606154453289d0703%2112000025045805357%21sea%21DE%210&curPageLogUid=GYWOkT0S87c3

Do you think this one will work as well. It's a development board so it's easier to wire up: https://www.alibaba.com/product-detail/Ebyte-E49-900M20S-test-board-Wireless_1600776754494.html