tbnobody/OpenDTU #421: Pinning change under settings [Request]

dAjaY85, 2022-12-09:
### Is your feature request related to a problem? Please describe.

Die D1 ESP32 Version ist vom Formfaktor sehr ähnlich zu der D1 Mini Version, dementsprechend ist es auch Möglich die ESP32 Version auf einen D1 mini Sockel drauf zu stecken. Leider passt dann aber die vordefinierte IRQ und CE Belegung des ESP32 Builds nicht dazu. Der IRQ Pin müsste auf GPIO17 und der CE Pin auf GPIO16 geändert werden. 

siehe Vergleich der Pin Belegung:
[image]

### Describe the solution you'd like

Es sollte eine Möglichkeit eingebracht werden, um die PIN-Belegung in den Einstellungen ändern zu können.

### Describe alternatives you've considered

Ein Zusätzliches ESP32 D1 Build mit der Komptabilität zum D1 Mini.

### Additional context

Durch diese Maßnahme wäre ein Wechsel vom D1 Mini ESP8266 auf einen D1 ESP32 mit geringen Aufwand möglich, es müsste lediglich der D1 Mini durch en D1 ESP32 getauscht werden.

tbnobody, 2023-01-17:
Dies  ist durch die Device Profiles nun möglich. Pin Belegungen können in json Dateien zusammengefasst, ausgetausch, hochgeladen und  ausgewählt werden.

closed 2023-01-17 by tbnobody

dAjaY85, 2023-01-18:
Benötige dafür mehr Informationen, bzw. Unterstützung bei der Display Klasse.