tbnobody/OpenDTU #424: Eveltlog - Einträge über mqtt

Verblizz, 2022-12-09:
### Describe the solution you'd like

OpenDTU sendet ja ne ganze Menge Daten über mqtt an meinen ioBroker, aber gibt es auch die Möglichkeit, die Statusmeldungen auch zu überttragen, vielleicht als html-Table o.Ä. ?
Ist sowas vorgesehen, oder gibt es das bereits und ich bin z ublind?

tbnobody, 2022-12-09:
ich hatte sowas auch schon im kopf. überlege seit längerem wie man das schön publishen könnte. eine html tabelle hat halt  den nachteil, dass man es nur schwer maschinell parsen kann. 
Um die letzte frage zu beantworten, geplant ja, umgesetzt aus oben genannten grund bisher noch nicht :)
Aber ggf. kann man im zuge von #317 hier noch etwas umbauen

kpwg, 2022-12-09:
Vorsichtige Frage: Welche "Statusmeldungen" fehlen Dir denn? Für ioBroker kann ich jetzt nicht sprechen, in FHEM habe ich im automatisch angelegten Device über MQTT so ziemlich alles, was man sich wünschen kann. Damit lassen sich dann entsprechend Daten aufbereiten und visualisieren.

Verblizz, 2022-12-10:
> Imho meinte er direkt das Eventlog (siehe Titel). Also das Fehlerlog usw...

Jupp, genau das meinte ich... Alle anderen Daten sind ja in ausreichender Anzahl vorhanden;)
Aber okay, dann warte ich einfach ab, ob es für dieses nicetohave irgendwann eine Lösung gibt.
Vielleicht kann man ja auch einfach die aktuelle / letzte Meldung vom Eventlog  publishen und mit Hilfe einer Zusatzsoftware (z.B. ioBroker) kann man sich die Einträge bei Bedarf auch in eine eigene Tabelle schreiben, oder halt immer nur den letzten Eintrag anzeigen (Darum ginge es mir hauptsächlich...)

CONSULitAS, 2022-12-14:
> > Vorsichtige Frage: Welche "Statusmeldungen" fehlen Dir denn? Für ioBroker kann ich jetzt nicht sprechen, in FHEM habe ich im automatisch angelegten Device über MQTT so ziemlich alles, was man sich wünschen kann. Damit lassen sich dann entsprechend Daten aufbereiten und visualisieren.
> 
> Imho meinte er direkt das Eventlog (siehe Titel). Also das Fehlerlog usw...

Dafür muss man dann aber nicht jedes mal das ganze Log übertragen.

Ich publishe Log-Ausgaben immer unter /.../meingerät/debug - Zeile für Zeile.

Dann kann man einerseits auf der Konsole mit `mosquitto_sub` sich eine Live-Anzeige holen, aber auch z.B. im FHEM die Ausgaben in ein Logfile schreiben.

Verblizz, 2022-12-14:
> Dafür muss man dann aber nicht jedes mal das ganze Log übertragen.
> 
> Ich publishe Log-Ausgaben immer unter /.../meingerät/debug - Zeile für Zeile.
> 
> Dann kann man einerseits auf der Konsole mit `mosquitto_sub` sich eine Live-Anzeige holen, aber auch z.B. im FHEM die Ausgaben in ein Logfile schreiben. Und im Gerät sieht man immer direkt die letzte Meldung.

Das stimmt, so hatte ich das ja auch angeregt, dass evtl. einfach nur die letzte Meldung vom OpenDTU published wird...

stefan123t, 2024-07-10:
Die Hoymiles DTU Pro stellt z.B. über Modbus RTU / TCP auch nur den letzten / aktuellste Alarm Code und den täglich bzw. nach einem Neustart stetig steigenden Alarm Count zur Verfügung:

Siehe z.B. `Technical-Note-Modbus-implementation-using-3Gen-DTU-Pro-V1.2.pdf`

### 5 Hoymiles Modus TCP Interface and Registers

#### 5.2 Modbus TCP Functions

##### 5.3.2 Microinverter Data Register List

| Registers | Name             | Decimal                | Units | Remark                                                       |
|-----------|------------------|------------------------|-------|--------------------------------------------------------------|
| 0x101C    | Alarm Code       | /                      | /     |                                                              |
| 0x101D    |                  |                        |       |                                                              |
| 0x101E    | Alarm Count      | /                      | /     |                                                              |
| 0x101F    |                  |                        |       |                                                              |