tbnobody/OpenDTU #432: Negative temperatures not displayed correctly

chbalnuweit, 2022-12-11:
### What happened?

Hello openDTU team,

Thank you for a great open source project!

I found a minor bug that I would like to report and bring to your attention:

If the measured temperature is below 0 °C  a value of 6553 + measured temperature is displayed in web interface and sent on MQTT by openDTU.

Example: 
actual temperature: -2 °C
displayed temperature: 6553 °C + (-2 °C) = 6551 °C

Would be great if you could implement a workaround :-)

Thanks!

### To Reproduce Bug

Wait until temperature drops below 0 °C and watch

### Expected Behavior

openDTU webinterface and MQTT is able to report negative temperatures

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2909325

MrReSc, 2022-12-11:
For me it was - 4°C this morning and that was displayed correctly.

closed 2022-12-11 by tbnobody

chbalnuweit, 2022-12-11:
sorry, my bad,  I was still using an older version. Thank you!