tbnobody/OpenDTU #450: DTU-Verbindung lässt sich "verbessern", indem man die Netzwerk-Settings abspeichert

Ascendor, 2022-12-13:
Hey @tbnobody,

have you seen this? I think this is worth analyzing:

> I noticed some strange behavior. I have connected the chip via Ethernet and it is not possible to connect to my inverters. As soon as I save the network / WLAN settings in the UI (no matter what is entered) a connection is possible for a few minutes. Then the connection breaks off again, until I save the WLAN settings again.

_Originally posted by @morzan1001 in https://github.com/tbnobody/OpenDTU/discussions/339#discussioncomment-4371785_
 
I can reproduce this 100%: My signal deteriorates over time. As soon as I press "save" on the network settings, it works rockstable again for a few minutes. Is this a hint at a bug in network hardware/software? 
Of does any kind of "reset" happen here, that could help if done on a regular (configurable?) basis automatically?

See my power graphs as a reference: 
The empty times are missing data. I pressed save at 14:14, 14:42 and 14:54. Afterwards, perfect line for 2 minutes...

tbnobody, 2022-12-13:
I have seen it and it doesn't make sense. The only thing what happens when the new wifi credentials are applied is a reconnect to the AP. This has nothing todo with the whole receiving logic. Only thing might be power spikes or something like this. I am running a lot of ESP instances with this software and it doesn't occour on any of them. So I would conclude that it's not a software issue.

Ascendor, 2022-12-13:
Thanks for the quick response. Weird thing. Note, it also works when not even changing the credentials. Don't know if it reconnect to the AP as well then.
Funnily enough, I don't get ANY connection while using ethernet - so there seems to be something with regard to wifi on this board (I also use an Olimex board)... 

I'll do some more experiments, when the sun is back ;)

morzan1001, 2022-12-14:
Hey guys, i fixed this issue. 

Heres what i did: 

I saw that there was a problem with setting the wifi mode during the upload/initial bootup.

```
[WiFiGeneric.cpp:1130] mode(): Could not set mode! 12308
```

I looked for the right place in the [code](https://github.com/tbnobody/OpenDTU/blob/9b0ab67849c4d93d6d7d96a34a1d6fcd3119bf65/src/NetworkSettings.cpp#L157) and saw that the WiFiMode is set to `WIFI_MODE_NULL". I tried to enter `WIFI_OFF` here, but that didn't change anything and I noticed later that `WIFI_MODE_NULL` points to `WIFI_OFF` via a define. 

Obviously, it is not possible to disable the Wifi module here or to change the mode. For this I stumbled over the following other [issue](https://github.com/espressif/arduino-esp32/issues/4842), but I have already checked, the fix is already included in the `WIFIGeneral.cpp` used here. 

I then noticed that a connection with my HM-300 module is always possible when the WIFI module is activated or in the status `WIFI_AP_STA`, this is set when you save the settings. In the UI, the WLAN module is then also displayed as Enabled. 

After a few minutes this setting is overwritten again, because the environment variable `OPENDTU_ETHERNET` is set or the `_networkMode` is set to Ethernet. 

The restart happens [here](https://github.com/tbnobody/OpenDTU/blob/9b0ab67849c4d93d6d7d96a34a1d6fcd3119bf65/src/NetworkSettings.cpp#L129), I think, because the admin mode is disabled after the 180sec timeout is reached here.

```
15:35:26.567 > Admin mode disabled
15:35:26.574 > WiFi disconnected
15:35:26.627 > E (184355) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb2700 (config or if_key is NULL or duplicate key)
15:35:26.638 > E (184357) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
15:35:26.644 > [184368][E][ETH.cpp:270] begin(): esp_eth_mac_new_esp32 failed
```

The WLAN module is now reactivated and it is no longer possible to connect to the HM-300.

But if you write `WIFI_MODE_STA` in both lines, 129 and 157, so that the WLAN module is switched on although you use ethernet, everything works without problems. 

But I would call this a workaround at most, because I actually want to turn off the WLAN module when I use Ethernet. Also, I really have no idea or found a connection between receiving data from the HM-300 and the WLAN module, I could imagine that the entire program flow is disturbed by the error when setting the WLAN mode, but this is only a vague assumption. 

An important side info maybe, it is not important if the chip is connected to any network. the main thing is that it is active. Also an access point does not have to be created.

[image]

tbnobody, 2022-12-14:
I will double check this at the weekend with my own hardware. Imho it was working without problems at @klahus1 
The Hoymiles lib does not contain any checks regarding connection state when receiving packages. The only check which is included is regarding the time. But in this case the no request packet is sent. But in your case, request packages are sent but you don't receive anything.

Ascendor, 2022-12-14:
> Hey guys, i fixed this issue.

Awesome, @morzan1001! Do you have a version/fork to test?

morzan1001, 2022-12-14:
> > Hey guys, i fixed this issue.
> 
> Awesome, @morzan1001! Do you have a version/fork to test?

No yet, i am hoping that @tbnobody will invastigate that further and maybe come up with a sollution 🥇 . For now you can just change `WIFI_MODE_NULL` to `WIFI_MODE_STA` [here](https://github.com/tbnobody/OpenDTU/blob/9b0ab67849c4d93d6d7d96a34a1d6fcd3119bf65/src/NetworkSettings.cpp#L129) and [here](https://github.com/tbnobody/OpenDTU/blob/9b0ab67849c4d93d6d7d96a34a1d6fcd3119bf65/src/NetworkSettings.cpp#L157).

tbnobody, 2022-12-14:
Was für ein Olimex Modul verwendet ihr genau? Und betreibt ihr es mit oder ohne POE?

Ascendor, 2022-12-14:
Ich nutze das Olimex ESP32-POE-ISO-EA-IND ohne POE.

Ascendor, 2022-12-15:
Vielleicht noch mal als Ergänzung/Klarstellung: 
Bei mir ist der Effekt auch da, wenn ich NICHT Ethernet, sondern WiFi verwende:

1. Perfekte DTU-Verbindung (ich kann jede Sekunde einen Messwert bekommen)
2. Nach exakt 180 Sekunden: DTU-Verbindung wird erheblich schlechter, nur alle paar Minuten kommen Daten rein 
3. Abspeichern der Netzwerk-Settings, weiterhin Wifi-Nutzung!
4. go to 1.

morzan1001, 2022-12-15:
[English version below]

@Ascendor wenn du dein Olimex Board sowieso mit WIFI benutzt, dann könntest du die Ethernet-Schnittstelle auch einfach deaktivieren. In der `platformio.ini` kannst du [hier](https://github.com/tbnobody/OpenDTU/blob/9b0ab67849c4d93d6d7d96a34a1d6fcd3119bf65/platformio.ini#L66) die Umgebungsvariable rausnehmen, dann sollte der WLAN Chip automatisch in den richtigen Mode gesetzt werden. 

@tbnobody Ich nutze das [Olimex ESP32-POE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) Board. Ich verwende Ethernet für eine Netzwerkverbindung aber kein POE, Strom kommt aktuell noch via USB. 

[Englisch version]

@Ascendor if you use your Olimex board with WIFI anyway, then you could just disable the Ethernet interface. In the `platformio.ini` you can take out the environment variable [here](https://github.com/tbnobody/OpenDTU/blob/9b0ab67849c4d93d6d7d96a34a1d6fcd3119bf65/platformio.ini#L66), then the WLAN chip should be set to the right mode automatically. 

@tbnobody I use the [Olimex ESP32-POE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) board. I use Ethernet for a network connection but no POE, power currently comes via USB.

Ascendor, 2022-12-15:
Ich nutze WiFi aktuell nur, weil es stabiler läuft. Ich möchte schon auf Ethernet setzen, wenn es geht.

tbnobody, 2022-12-17:
I've just created a small test installation:
* ESP32-POE-ISO Board
* Power via USB
* Ethernet connected
* Cheap NRF24 module (https://www.amazon.de/gp/product/B07VQ838KT/)
* RF Power is at Minimum (-18dbm) 
* Module is located in the cellar, inverter somewhere in the garden

--> No Problems in ethernet mode

After the grace period the internal wifi will be disabled as well, and I am also seeing these warning:
```
Admin mode disabled
E (534789) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffb2700 (config or if_key is NULL or duplicate key)
E (534791) esp.emac: esp_eth_mac_new_esp32(595): alloc emac interrupt failed
[529860][E][ETH.cpp:270] begin(): esp_eth_mac_new_esp32 failed
```

But there is no effect on the function.

I would guess it's an hardware issue on your side. Either the power supply is not sufficient or creates ripples or something different. It may work under some cercumstances but seems to fail when e.g. more or less power is drawn (by changing the wifi mode e.g.).

I  know that is no satisfying answer, but it's working here so I would guess no issue in the source and ESP framework.

Another question:
* What's your current RF power level?
* Does your ethernet network and your wifi network belong to the same subnet?
* What is your current poll interval? I would not set it below the default of 5 seconds as it will maybe block some other actions (and you wrote you receive data every second so you must have set this to a lower value)

Ascendor, 2022-12-17:
Thank you so much for taking the time and effort to test it.
I agree it's most probably a very specific hardware issue.

> * What's your current RF power level?
Minimum / -18, but I can reproduce it at any Level

> * Does your ethernet network and your wifi network belong to the same subnet?
yes

> * What is your current poll interval? I would not set it below the default of 5 seconds as it will maybe block some other actions (and you wrote you receive data every second so you must have set this to a lower value)
I've only set it to 1s in order to make the issue more visible. When clicking save you can see the live view to update every second right away. With 5 seconds Interval, you have to wait 10-20 seconds to be sure, because a single request can always fail I guess. 

But, as above: It can also bei reproduced with every other Interval setting

morzan1001, 2022-12-17:
> --> No Problems in ethernet mode

@tbnobody do you get also the warning `Could not set mode! 12308` on startup?

tbnobody, 2022-12-17:
> @tbnobody do you get also the warning `Could not set mode! 12308` on startup?

Nope... I tried it several times now (pressed the reset button) and always got the following output:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Initialize Network... ETH start
ETH connected
done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize MqTT... done
Initialize WebApi... done
Check for default DTU serial... done
Initialize Hoymiles interface... E (4275) wifi:Set status to INIT
WiFi disconnected
Connection successfull
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
  Adding inverter: 116171603546 - Meine Solaranlage done
done
Switch to Ethernet mode
Configuring Ethernet DHCP IP... done
Setting Hostname... done
ETH got IP: 192.168.1.119Network connected
Connecting to MQTT...
Connected to MQTT.
```

Ascendor, 2022-12-17:
> @Ascendor if you use your Olimex board with WIFI anyway, then you could just disable the Ethernet interface. In the `platformio.ini` you can take out the environment variable [...]

I've tried this today. When I outcomment or remove the line, then I don't get a DTU connection running at all anymore. This is so strange.

Ascendor, 2023-01-24:
I have updated to the newest firmware yesterday, and for me everything works flawlessly now. I can even get a DTU connect with ethernet connected now. :)

The only relevant network related commits I see are: 

- https://github.com/tbnobody/OpenDTU/commit/3d3feb3517b4614c2d52e439bd77e20270ad125e
- https://github.com/tbnobody/OpenDTU/commit/c85b99ffc1fd0870e05988037d3879bfd910b7a2 

No idea, if/how they can be relevant here.

Yet, as my problem seems resolved, I am closing this issue.

closed 2023-01-24 by Ascendor

Ascendor, 2023-01-24:
Add: A difference I see, is that the ESP is now connected via Ethernet AND WiFi. Before the patch, the Wifi module deactivated when using Ethernet. Maybe that makes some kind of difference.

ude6, 2023-03-10:
I actually just compiled fro GIT today and had the same issue as described here. 
I actually observed that when I used the git master, that WIFI did disable when ETH was active.
I then made the changes @morzan1001 described and the code seems to run fine (i.e. WIFI is enabled)

lke15, 2023-04-30:
Hi, i had the same error. When i use the olimex esp32-poe-iso with wifi it works fine. So when I use ethernet and poe or ethernet without poe and 5v usb, opendtu receive few minutes from hoymiles and then stops. I can go to the network settings options and click save, then it receive data again for few minutes.

Gerd-Watzmann, 2023-11-03:
Hi, I habe exactly the same problem with my newly configured Olimex ESP32 POE ISO board and the firmware 23.10.9

It all works perfectly for a few minutes after having pushed the "save" button of the network configuration (without having made any changes, and empty WiFi credentials). Then both inverters go offline again, until the next "save".

I switched from AhoyDTU, since I was not able to connect Ahoy via Ethernet (W5500 module). Long story. So everybody recommended to switch to OpenDTU, since Ethernet ist not a problem with OpenDTU.  Yes, Ethernet works - but unfortunately now I have this new problem :-(

Could there be a solution to this? WiFi is not always available were I plan to install the DTU. But I have a LAN port there, so I would like to use this.

Many thanks in advance!!