tbnobody/OpenDTU #451: CMD Reiter im Iobroker fehlt

FloW-80, 2022-12-15:
### What happened?

Bei mir sind im Iobroker unter Objekte - mqtt - solar und dann der jeweilige Inverter nur 0,1,2,3,4, Device und Status vorhanden,  CMD fehlt. Ich hatte letzte Woche erst die DTU auf die neueste Version gebracht.

### To Reproduce Bug

-

### Expected Behavior

-

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

9bfc704

tbnobody, 2022-12-15:
Ich weiß nicht was ioBroker als Tabs darstellt. Ich vermute das sind die verschiedenen Topics die ioBroker in der Vergangenheit schon einmal gesehen hat. Wenn du noch nichts an eine Topic geschickt hast wird die dort auch nicht auftauchen.

JauntyJosef, 2022-12-17:
Genau so ist es. Wenn du noch nie was an das Topic cmd geschickt hast, dann siehst du im ioBroker auch nichts.

Dodger2022, 2022-12-19:
Hallo,
ich muss hier nochmal genau nachfragen, da ich das gleiche "Problem" habe.

Ich versuche seit Tagen diese Topics zum laufen zu bringen damit ich die Max-Leistung über IObroker steuern kann.
(soll mal eine Nacheinspeisung von einem extern Batteriesystem werden an einem freien Eingang an einem HM-600)

Ich habe versuche über den MQTT-Explorer die Topics anzusprechen und den HM600 zu regeln. 
Ich sehe auch im Explorer die Meldungen Name;Device;status;0;1;2 usw mit ihren Inhalten.
Wenn ich unter Puplish/Topic etwas eintrage und puplishe legt IOBroker diese Einträge auch an, aber der Hoymiles setzt es nicht um oder schaltet ab (Statusleiste wird gelb) oder setzt Current Limit auf 0W/%.
[image]

(zb : OpenDTU/serialXXXXXXXserial/cmd/limit_nonpersistent_relative/50% ) 
       
Oder ist nur die Schreibweise falsch?

Interessant finde ich auch, das wenn ich OpenDTU neu starte tauchen im MQTT-Log des IOBrokers die CMD-Topics kurz auf, werden aber in der Objektliste nicht angezeigt/angelegt.
[image]

Sorry wenn ich mich doof anstelle aber muss mich mit MQTT noch vertraut machen.

tbnobody, 2022-12-19:
> (zb : OpenDTU/serialXXXXXXXserial/cmd/limit_nonpersistent_relative/50% )

Das ist hier ist falsch
Bei Topic muss `OpenDTU/serialXXXXXXXserial/cmd/limit_nonpersistent_relative` stehen. Der Wert (Eingabefeld darunter im MQTT Explorer) muss `50` sein.

romew137, 2022-12-19:
https://github.com/tbnobody/OpenDTU/issues/35
Lest euch mal durch, was Hubsi5 am 14 Oktober geschrieben hat.
Hatte das selbe Problem und habe es damit behoben bekommen.

tbnobody, 2022-12-19:
Ich verschiebe das mal in die Diskussionen weil es kein Issue von OpenDTU ist.