tbnobody/OpenDTU #461: Nachteinspeisung/Tageinspeisung

romew137, 2022-12-19:
### What happened?

Ich habe zwei Wechselrichter, einen für Tagsüber (HM-1500) und einen für Nachts (HM-600). Der Wechselrichter für Nachts wurde nicht gefunden, erst wo ich den für tagsüber gelöscht habe (dieser hatte durch Mangel an Sonnenlicht keine Verbindung mehr), hat der andere eine Verbindung hergestellt. Solange ich den Wechselrichter für tagsüber gelöscht lasse, habe ich keine Probleme mit der Verbindung. Sobald ich diesen wieder hinzufüge, ist die Verbindung zum HM-600 sehr sporadisch und bricht öfters ab. Das Problem hatte ich den ganzen Tag mit dem HM-1500 nicht. Man muss auch dazu sagen, das ich den HM-600 erst heute Abend eingefügt habe, als der HM-1500 schon offline war. Ich habe auch eine weitere DTU hier liegen mit Ahoy installiert, da funktionieren beide ohne Probleme (Tag und Nacht).
Zu den technischen Daten:
- AZDelivery ESP32 NodeMCU Modul Dev Kit C
- NRF24L01+ - PA+LNA SMA
- NRF24L01+ PCB Adapter

[image]
[image]
[image]
[image]

### To Reproduce Bug

Wenn einer von zwei Wechselrichtern offline ist.

### Expected Behavior

Ich kenne nicht den Unterschied zur Ahoy DTU, nur hat diese anscheinend weniger ein Problem damit.
Vielleicht liegt das Problem an der Abfrage?

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

3421c9a

### Relevant log/trace output

```Shell
Ich weiß leider nicht, wo ich sowas bei der OpenDTU finde.
```

tbnobody, 2022-12-19:
> Wenn einer von zwei Wechselrichtern offline ist.

Kann so ansich nicht sein. Ich  habe hier standardmäßig 2 Wechselrichter eingetragen. Einen den ich wirklich besitze und einen mit einer zufälligen Seriennummer den ich einfach eingetragen habe um bei mehreren Invertern auf keine Fehler zu stoßen. Bisher lief das alles absolut problemlos.

Generell ist es auch so,  dass hier keine "feste" Verbindung zu einem Inverter aufgebaut wird. Man sendet Anfragepakete und entweder der Inverter antwortet oder auch nicht.

romew137, 2022-12-19:
Ich habe noch eine DTU Pro hier stehen, welche zurzeit nicht in Betrieb ist.

D3R-ST3FAN, 2022-12-21:
>für Nachts (HM-600)

Ich weiß, off-toppic, aber wie genau nutzt Du den HM-600 für die Nacht?
Direkt ne Batterie dran? Spinnt dann nicht der MPPT?

romew137, 2022-12-21:
12,8V LiFePo4 100Ah mit DC-DC Wandler (Step Up Modul).
Output ca. 48V 6A.

romew137, 2022-12-21:
Ich würde das Thema hier auch erstmal wieder schließen, weil ich das Problem seit 2 Tagen nicht mehr reproduzieren kann.
Ich vermute, dass es vielleicht wirklich am Empfang gelegen hat. OpenDTU läuft jetzt seit 2 Tagen stabil mit beiden Wechselrichtern. Trotzdem erstmal Danke für die Hilfe und Danke für die super Arbeit.

closed 2022-12-21 by romew137