tbnobody/OpenDTU #464: KiCad Data not readable

erniberni, 2022-12-20:
### What happened?

I tried to load the KiCad project to KiCad V. 5.1.4
The project did not load and every line throws the error
"20:29:49: file './Projekte/opendtu/openDTU-BreakoutBoard-main/openDTU.pro', line 1: '=' expected."
The format of the file is not as expected.

### To Reproduce Bug

loading in KiCad 5.1.4

### Expected Behavior

project is readable

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

.

### Relevant log/trace output

```Shell
.
```

### Anything else?

.

tbnobody, 2022-12-20:
Where do you see a KiCAD project file? I am not aware that I added one  to this repository

erniberni, 2022-12-20:
ups, sorry - wrong git account
breakout board is on 
https://GitHub.com/dokuhn/openDTU-BreakoutBoard
installed KiCad 6 and project opened

closed 2022-12-20 by tbnobody