tbnobody/OpenDTU #470: [Request] Zero Werte nicht per MQTT zu übertragen

MiniOh, 2022-12-26:
### Describe the solution you'd like

Hallo, 

in dem Projekt Ahoy-DTU gibt es die Möglichkeit, "falsche" Daten, z.B. Zero Werte nicht per MQTT zu übertragen.
Durch den Parallel Betrieb mit einer HM-DTU kommt es ja leider vor, dass, alle paar Minuten die falsche Werte geloggt werden, z.B. 0 beim /powerdc oder auch extrem hohe Werte beim /yieldday.
Es wäre schön, wenn man diese Werte filtern könnte und somit nicht per MQTT zu übertragen? Wäre das möglich?

Vielen Dank schon vorab.

===========================================================

Hello,

In the Ahoy-DTU project, there is the option of not transmitting "wrong" data, e.g. zero values, via MQTT.
Unfortunately, parallel operation with an HM-DTU means that the wrong values are logged every few minutes, e.g. 0 for /powerdc or extremely high values for /yieldday.
It would be nice if you could filter these values and not transfer them via MQTT? Would it be possible?

Thank you in advance.

tbnobody, 2022-12-26:
Duplicate of #26

closed 2022-12-26 by tbnobody

MiniOh, 2022-12-26:
Gibt es Pläne, hier eine Anpassung vorzunehmen?