tbnobody/OpenDTU #475: Habe versucht eine Version mit 24 Invertern zu übersetzen. Klappt nicht - könnt Ihr helfen ?[Request]

maischrot, 2022-12-26:
### Is your feature request related to a problem? Please describe.

tried to change Max Inverter Count to 24. does not work

### Describe the solution you'd like

mehr inverter

### Describe alternatives you've considered

d

tbnobody, 2022-12-27:
Do you get an error message when adding the 15th inverter?

maischrot, 2022-12-29:
Hi, i get no error messages. i can add all 23 inverters i need. It seems the define is not recognized in the Java Scripts for the GUI, since only 13 inverters are displayed in „settings/inverters“ (14th is partially displayed). In the data export, i can find 12 inverterd, but in the live view i can see 18 of 24. I reduced the  Size of strings to keep the setup structure small, but without success.

tbnobody, 2022-12-29:
> since only 13 inverters are displayed in

Does the API endpoint `/api/inverter/list` return all inverters? If not it's not a issue with the web interface (I don't think thats its a issue with the web ui).
Did you doublecheck the `maxJsonBufferSize` of all the `AsyncJsonResponse` objects?

maischrot, 2022-12-29:
it helped to increase response size in webApi_inverter.cpp

Which additional values should be changed for „live data“ (max 18) and saving of inverters?

maischrot, 2022-12-29:
I additionally changed the other appearances, which now displays 23 inverterd in the web app, but still does not store more than 12 after reboot

tbnobody, 2022-12-31:
> I additionally changed the other appearances

That means you also adjusted the config reading and parsing methods?

maischrot, 2022-12-31:
No, i would be happy to get hints where to change stuff. thanks

maischrot, 2023-01-06:
Hello,
My goal is to display 25 inverters in the system
 
I changed the length of the JSON Responses, as suggested to parse the longer inverter list. 
The inverters are correctly displayed in the /api/inverter/list

One Problem with liveView remains:
- in Safari the live View list is limited to 19 Entries.
- in Firefox the inverters appear one time, until the display is updated, then the emitter list is limited to 19 devices.

Thank you for hints.

-------------------------------

Here is what I did and where I modified the code:

config.h:  
#define INV_MAX_COUNT 24
#define JSON_BUFFER_SIZE 3*6144
#define CHAN_MAX_NAME_STRLEN 11

webApi_mqtt.h
#define MQTT_JSON_DOC_SIZE 2*3072

web_api_ws_live.cpp:
- #define MAX_LIVE_SIZE 70000
used in Line 217: AsyncJsonResponse* response = new AsyncJsonResponse(false, MAX_LIVE_SIZE);

I did comparable stuff in the other Json Response calls:
WebApi_config.cpp, 
Webapi_devInfo.cpp
,WebApi_dtu.cpp,
WebApi_eventlog.cpp,
WebApi_inverter.cpp,
webApi_limit.cpp,
webApi_power.cpp
   
 AsyncJsonResponse* response = new AsyncJsonResponse(false, 3*4096U);

tbnobody, 2023-05-08:
It is hard enough to get enough contiguous memory for 10 inverters. Even there, there are sometimes problems assembling the json responses. I will therefore not support more than 10. In exceptional cases, you can also use several DTUs.

closed 2023-05-08 by tbnobody