tbnobody/OpenDTU #486: Unknown model

vgr28, 2023-01-02:
### What happened?

Hardware Teilenummer: 132449357
Modell: HM-800

### To Reproduce Bug

On Firmware: dfec263

### Expected Behavior

Show Modell

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

dfec263

tbnobody, 2023-01-02:
Based on the serial number it looks like an HMS-800? Could you doublecheck this?

vgr28, 2023-01-02:
Could be, what is the different between HM-800 and HMS-800? On Firmeware before it was shown as HM-800 and the HArdware Teilenummer 269565952. Also interesting that the HW-number is also changed in latest Firmeware https://github.com/tbnobody/OpenDTU/commit/dfec263d65826f30e4c311f3cf019c033c26b820.

tbnobody, 2023-01-02:
> Could be, what is the different between HM-800 and HMS-800?

Different communication frequency, different RF Module, maybe different protocol?

> On Firmeware before it was shown as HM-800 and the HArdware Teilenummer 269565952

From when was your previous firmware? I didn't change anything regarding the the device detection for several month.
On the other hand, your serial number looks much to short. Inverter serial numbers have 12 digits

tbnobody, 2023-01-02:
Sorry.... My fault.... You are refering the to the "Inverter Info" dialog aren't you?

vgr28, 2023-01-02:
after back and forward flashing it's ok. but behavior never seen before..
[image]

tbnobody, 2023-01-02:
hm I would close this issue as it's working now and therefor its a little bit hard to reproduce.

closed 2023-01-02 by tbnobody