tbnobody/OpenDTU #495: Night feed / Total power display

fuxe13, 2023-01-05:
### What happened?

I use an inverter for night feeding. This is switched by a relay. After switching off, the values remain displayed and are added to the total output. This would have to be taken into account somehow in order to get a correct value.

### To Reproduce Bug

I use an inverter for night feeding. This is switched by a relay. After switching off, the values remain displayed and are added to the total output. This would have to be taken into account somehow in order to get a correct value.

### Expected Behavior

After switching off, the values remain displayed and are added to the total output. This would have to be taken into account somehow in order to get a correct value.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

64afbf8

tbnobody, 2023-01-05:
I am not sure if I understand you correct of whether this is a bug. The total yield value can not be modified as it comes directly out of the inverter and is a always growing value which cannot be reset.

fuxe13, 2023-01-06:
Re: [tbnobody/OpenDTU] Night feed / Total power display (Issue #495)

Hi!

The "problem" is that with the inverter for night feed-in, the values remain at e.g. 250 watts and this is calculated throughout the day, although the inverter is no longer connected to the grid.
I hope I was able to explain it in a reasonably understandable way.

am Donnerstag, 5. Jänner 2023 um 11:11 schrieben Sie:

I am not sure if I understand you correct of whether this is a bug. The total yield value can not be modified as it comes directly out of the inverter and is a always growing value which cannot be reset.
—

Mit freundlichen Grüßen
Walter Klammer

tbnobody, 2024-01-20:
This should be fixed if you enable the "Zero runtime data" option for the inverter.

closed 2024-01-20 by tbnobody