tbnobody/OpenDTU #496: zero export via Home Assistent - opendtu lost connection to HM 1500

mag2bue, 2023-01-07:
### What happened?

If I update the power frequently, opendtu lost the connection to HM 1500. After restart opendtu, connection is back.

### To Reproduce Bug

Update the power each 5s

### Expected Behavior

Opendtu lost the connection to HM1500. 
Restart Opendtu -> connection are back.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

64afbf8

### Anything else?

I use Home Assistent via a automation to do "zero export"  If the automation runs, after around 10min, opendtu lost the connection to hm1500. I was hoping the connectin comes back - after 1h I did a restat of opendtu, connection come back.

INFO: update the power is via number.xxxx_limit_persistent_absolute

tbnobody, 2023-01-08:
What shows the serial console?

There is NO static connection between the DTU and the inverter. For every request it is as it would be the first one. 
Have you tried to send the requests in a longer period than 5s?
Not sure if this short period is working.

LJ9210, 2023-01-08:
@mag2bue  how did you enable zero export via automation?  can you put the yaml code here?

mag2bue, 2023-01-08:
Good to know. Thanks! 

I will change my update time to 30s.
After the next sunny day, I will come back.

serial console?   where can I  find it?

Archibald-Harry-Tuttle, 2023-01-08:
OpenDTU > info > console

mag2bue, 2023-01-09:
funny - I can see the console on my mobile phone. but no on my computer.

[image]

Browser: Chrome, Windows11

tbnobody, 2023-01-09:
You should either clean your cache or reload the page with ctrl+f5. That should fix it

tbnobody, 2023-01-30:
I assume that a increase of the update time solved the issue. Will close it... feel free to open it again if new information is available.

closed 2023-01-30 by tbnobody