tbnobody/OpenDTU #505: OpenDtu not functioning with some ESP32 Boards

saschTa, 2023-01-09:
### What happened?

Built the code from source (Master) today and flashed it to this board:

https://www.makershop.de/plattformen/nodemcu/esp32-board-stiftleisten/

it consistently did not allow me to connect to the AP OpenDtu-*
I tried it with all devices I have (Macbook Pro, Android, Windows)

Log output seemed absolutely normal  (no Error) and AP was created successfully. Just connecting via wifi is not possible.
I erased wifi and flash multiple times -> no Luck
I thought the ESP was broken.

Tried a little wifi ap programm -> could connect without issues
Tried ahoyDTU -> no issues on connection with all devices while it was in AP mode

### To Reproduce Bug

Use above mentioned board, to make AP mode fail in that no device can connect to ESP32

### Expected Behavior

Working normally

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

64afbf8

tbnobody, 2023-01-09:
As you can maybe imagine its a little bit hard for me to fix such a issue. I have the software running with multiple ESP boards without any problem. What shows the serial console?

tbnobody, 2023-01-09:
You can also try to perform a full erase of the ESP. This will also clear some previously set wifi credentials etc. 
Can be found in the Projects tasks (Erase Flash):
[image]

saschTa, 2023-01-09:
I did the flash erase before any flashing action just to make sure. What now makes me completely irritaded, is that after switching from ahoy to OpenDtu without an erase in between, everything works fine. 
I will close this issue for now, since I need to get it reproducible again/

Log was exactly this when connecting did fail:
```
00:27:11.526 > E (566) esp_core_dump_flash: No core dum����ѥѥ���found!
00:27:11.526 > E (566) esp_core_dump_flash: No core dump partition found!
00:27:11.526 > 
00:27:11.526 > Starting OpenDTU
00:27:11.526 > Initialize FS... done
00:27:11.526 > Reading configuration... done
00:27:11.526 > Initialize Network... done
00:27:11.526 > Setting Hostname... Initialize NTP... done
00:27:11.526 > Initialize MqTT... done
00:27:11.526 > Initialize WebApi... done
00:27:11.526 > Check for default DTU serial... done
00:27:11.526 > Initialize Hoymiles interface... Connection successfull
00:27:11.526 >   Setting radio PA level... 
00:27:11.526 >   Setting DTU serial... 
00:27:11.526 >   Setting poll interval... 
00:27:11.526 > done
00:27:11.526 > Switch to WiFi mode
00:27:11.526 > Setting Hostname... done
```

saschTa, 2023-01-09:
Update: Got the error: It is important to cut power once after flashing for some reason. Just resetting via Pin seems not enough. When I cut power after fresh install and then connect, everything works fine.

I am sorry for causing additional load :).

closed 2023-01-09 by saschTa

delfmannes, 2023-04-17:
Hello saschTa,
I have the same board. Could you send me your settings,
like...
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=13
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5