tbnobody/OpenDTU #507: compatible with solenso SOL-H400 ??

dimgod59, 2023-01-13:
### Is your feature request related to a problem? Please describe.

Hello,
Do you know if OPEN DTU works with SOLENSO type SOL-H400 micro inverters?
and can it work at the same time with a HOYMILLES 400 micro inverter?
Thanking you in advance.

Best.

Bonjour, 
Savez vous si OPEN DTU fonctionne avec les micro onduleur de la marque SOLENSO type SOL-H400 ? 
et peut il fonctionner en meme temps avec un micro onduleur HOYMILLES 400 ?
Vous remerciant par avance.

Cordialement.

### Describe the solution you'd like

yes or no or alternative OPEN DTU :)

### Additional context

i have already micro inverters HOYMILLES 400

tbnobody, 2023-01-13:
could be that it's working. could you tell me the first (from left to right) 4 digits of the serial number?

dimgod59, 2023-01-13:
I do not yet have a panel with a Solenso micro inverter.  the first 4 digits correspond to the month and year of manufacture?  I have a friend who has one on a photo I find type numbers 1121xxxxxx can this be enough?