tbnobody/OpenDTU #509: opendtu works with my Hoymiles HM-1500. I have also a Hoymiles MI-600. What can I do for enabling with DTU ?

wgbartels, 2023-01-13:
### Describe the solution you'd like

Please enable openDTU for MI-600

wgbartels, 2023-01-20:
Unfortunately I am only a user, not an expert. How can I implement the gen2 protocol and where can I get it? Can you help me?

stefan123t, 2023-01-24:
Implementation based on an old AhoyDTU Fork by Ziyat T is here https://github.com/Ziyatoe/DTUsimMI-Hoymiles

Gen2 Status and Data Commands are 0x08..0x11 for MI-300/600 and 0x36..0x39 for MI-1000..1200

stefan123t, 2023-11-22:
Maybe not a solution on OpenDTU side, but rejoe2 implemented the MI-300..1200 support in the Ahoy DTU master code.
E.g. here https://github.com/lumapu/ahoy/issues/1111 some additional model details.

stefan123t, 2024-07-10:
@tbnobody is this still planned to port the changes made by @rejoe2 in Ahoy to OpenDTU in order to also support the older Hoymiles MI-300..1200 models in OpenDTU ?