tbnobody/OpenDTU #513: MQTT stops at inverter shutdown and does not start again when inverter is back up

uis0815, 2023-01-14:
### What happened?

After upgrading to the newest firmware (as of Jan. 14th 2032) mqtt information is sent as long as the inverter runs. After inverter shutdown in the evening mqtt stops sending data. When the inverter starts at the next morning, mqtt does not again start transmitting data.

### To Reproduce Bug

wait for one day / inverter off-on cycle

### Expected Behavior

start transmitting mqtt data when inverter is back up again / never stop sending mqtt data.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

ee79799

### Anything else?

Mqtt worked like a charm with older firmware versions (from beginning December 2022). To restart mqtt, either restart esp or go to the mqtt settings page (toggle enable to off and to on again) and hit "Save" button.

tbnobody, 2023-01-14:
I did not change anything related to the mqtt behavior: https://github.com/tbnobody/OpenDTU/compare/9bfc7044fcd3d4853c8519cb93534ece577e37ed..HEAD

I am getting values at night as well. I've doublecheched it here several times --> cannot reproduce it

uis0815, 2023-01-18:
Has not happened again after toggling mqtt twice and saving configuration on Jan 14th. Thank you for the outstanding work and the great project

closed 2023-01-18 by uis0815