tbnobody/OpenDTU #519: NO contact to TSOL M1600 with serial 12*

MuckelMarko, 2023-01-17:
### Is your feature request related to a problem? Please describe.

Hi what a cool reverse engineering project. 
Unfortunately I got a TSOL M1600 Micro Inverter with serial number 12 *. 
There is no communication between TSOL and openDTU. See console logging in attached file (console.txt).  
Does someone have a idea hast wrong? 
The Serial contains characters (TSOL-serial.jpg). Is this the problem? 

[console.txt](https://github.com/tbnobody/OpenDTU/files/10434035/console.txt)
[image]
thank you for this cool project!

### Describe the solution you'd like

Is there a possibility to update OpenDTU to be able to communicate with a TSOL M1660 with probably new firmware?

tbnobody, 2023-01-17:
If you look at the inverter settings. I would assume your inverter was detected as HM-1500 etc?
It is currently sending packets. That means that the serial number is known for a specific device. Is the NRF Module correctly detected? (Info --> System)

MuckelMarko, 2023-01-18:
Due the Installation it tells me someting about HM-1500 Type. I will check it. 

The NRF Module seems correct detected: 
<html><body>
<div class="card-header text-bg-primary">Radio Information</div><div class="card-body"><div class="table-responsive">

Chip Status | connected
-- | --
nRF24L01+

</div></div>
</body>
</html>

UweSchneiderGmbH, 2023-03-06:
Hmm,

I've a TSOL with a serial number beginning with 11.. but my serial number sticker looks complete different (sorry for the bad quality of the picture):

[image]

My difficulty with the connection was, that the TSOL needs a connected solar panel (or DC source) with enough power (~10W) before the inverter activates and is answering to the connection requests. With less power the LED blinks red and green, but no communication. With only AC connected the inverter does nothing.

The inverter was bought in January 2023 and is reported as HM-1500. The bootloader version is 0.1.0 and the firmware is 1.0.12 with the date 2020-06-24.

stefan123t, 2023-11-22:
TSOL and some OEM / so called SubBrands of Hoymiles may require a different Serial Number calculation which rovo89 found out on [Discord](https://discord.com/channels/984173303147155506/1146388875204378636/1168573556163100813).

> rovo89: schau dir ggf. mal `ProductSNGeneUtils.sn92long()` an.
```java
public static long sn92long(String str) {
  if (str.length() != 9) {
    throw new IllegalArgumentException("Illegal SN");
  }
  int[] iArr = _FASTDICT;
  return ((((iArr[str.charAt(0)] << 10) | (iArr[str.charAt(1)] << 5)) | iArr[str.charAt(2)]) << 32)
          | iArr[str.charAt(8)] | (iArr[str.charAt(3)] << 25) | (iArr[str.charAt(4)] << 20)
          | (iArr[str.charAt(5)] << 15) | (iArr[str.charAt(6)] << 10) | (iArr[str.charAt(7)] << 5);
}
```

> Die 9 Stellen dürften die vor dem Bindestrich sein, also `A11001TW3`, die `193` ist ein "Extend code" (Prüfnummer?).
`_FASTDICT` ist ein Mapping für die möglichen Zeichen zu ihrer Position in folgender Liste: `0123456789ABCDEFGHJKLMNPRSTUVWXY`
> 
> Da fehlen ein paar Zeichen, die leicht zu verwechseln sind: I, O, Q, Z. So ergeben sich 32 Zeichen, d.h. 0 = 0, A = 10, Y = 31
Mit dem Code oben wird ein 64-Bit-Wert erzeugt. Da der größte Wert 31 ist, passt er in 5 Bit, und so ergeben die Bitverschiebungen mehr Sinn. Wenn man Seriennummer als `ABCDEFGHI` darstellt, finden sich die jeweiligen Werte nachher so wieder: `0AAA AABB BBBC CCCC 00DD DDDE EEEE FFFF FGGG GGHH HHHI IIII`. Jetzt könnte man es von Hand ausrechnen, aber ich würde es lieber heute Abend mal in ein Python-Skript packen und damit ausrechnen. Kontrollmöglichkeiten gibt es vermutlich nicht?
Wenn man das Ergebnis in Hex darstellt, ergibt sich auf jeden Fall eine 12-stellige Zahl, was zu den klassischen Seriennummern passt.

```python
sn = 'A11001TW4'
CHARS32 = '0123456789ABCDEFGHJKLMNPRSTUVWXY'
CHARS64 = 'HMFLGW5XC301234567899Z67YRT2S8ABCDEFGHJKDVEJ4KQPUALMNPRSTUVWXYNB'

first_char = '1'
i = CHARS32.find(first_char)
sum1 = sum(ord(c) for c in sn) & 31
sum2 = sum(CHARS32.find(c) for c in sn) & 31
ext = first_char + CHARS64[sum1 + i] + CHARS64[sum2 + i]
print(ext)
```

> rovo89: Übrigens sind die ersten vier Stellen wohl der Gerätetyp (A=MI, B=ES, C=RSD, D=DTU), dann zwei Stellen für den OEM, dann eine Stelle für die Fabriknummer. Die restlichen 5 Stellen sind eine Sequenznummer. Plus dann halt ein Bindestrich und der "Extend Code".
> Für die "klassischen", 12-stelligen Seriennummern gibt es übrigens auch einen Extend Code:

```python
sum1 = sum(ord(c) for c in sn) & 31
sum2 = sum(CHARS32.find(c) for c in sn) & 31
ext = CHARS64[sum1] + CHARS64[sum2]
print(ext)
```

Ich glaube das Prefix der Seriennummer entscheidet idR noch darüber über welches RF-Modul (NRF24L01+, CMT2300A und in Zukunft zweites WLAN Interface für WiFi-Series) die OpenDTU/AhoyDTU mit dem WR (bzw in Zukunft dessen DTUBI) kommuniziert.

Wie rovo89 richtig vermutet wird das Prefix 4412 bzw 2821 mW noch nicht von der OpenDTU erwartet / akzeptiert bzw der bisherige Code weiss nicht was er mit dem WR anfangen soll. Da fehlen bestimmt noch ein paar Anpassungen für die **HERF** und andere “SubBrand” Inverter.

Birkenstab, 2024-02-14:
Is there any information on whether OpenDTU supports the Herf-800 inverter or whether somebody has already figured some stuff out on how to support it?

Birkenstab, 2024-02-15:
Nevermind I figured something out: https://github.com/tbnobody/OpenDTU/issues/1751

tbnobody, 2024-02-15:
I also added some remarks etc. in #1751

tbnobody, 2024-02-15:
But as far as I see it, it's still unknown how to decode the `RE12508005262`?

stefan123t, 2024-07-10:
@tbnobody is there an issue with the above code for sn92long() decoding the `RE12508005262` ?

I did try that Serial ID with the above sn92long code from https://github.com/tbnobody/OpenDTU/issues/1751#issue-2137109935 and https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1946966422

SN92 Long: `RE12508005262-1X5`
Serial ID: `61C104502000`

Whom is this Serial Number from and does it work with the Serial ID above, e.g. 114104502000 or similar with a different 1-in1, 2-in1 or 4-in-1 mode serial prefix ?

stefan123t, 2024-07-10:
@MuckelMarko I also tried to scan / detect the bar code given in your image, though it is not sharp enough to yield a tangible result. 

May I ask you to make a better mugshot of your inverter Serial ID ?

[image]

```bash
zbarimg HERF-M1600_barcode_levels.jpg 
scanned 0 barcode symbols from 1 images in 0,01 seconds
```