tbnobody/OpenDTU #520: SSL-Certificate must not be longer than 2048 chars - why?

eingemaischt, 2023-01-17:
### What happened?

I want to connect opendtu to myqtthub.com - they are using sectigo as certificate provider.

This is their root certificate:

-----BEGIN CERTIFICATE-----
MIIF3jCCA8agAwIBAgIQAf1tMPyjylGoG7xkDjUDLTANBgkqhkiG9w0BAQwFADCB
iDELMAkGA1UEBhMCVVMxEzARBgNVBAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0pl
cnNleSBDaXR5MR4wHAYDVQQKExVUaGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNV
BAMTJVVTRVJUcnVzdCBSU0EgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMTAw
MjAxMDAwMDAwWhcNMzgwMTE4MjM1OTU5WjCBiDELMAkGA1UEBhMCVVMxEzARBgNV
BAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0plcnNleSBDaXR5MR4wHAYDVQQKExVU
aGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNVBAMTJVVTRVJUcnVzdCBSU0EgQ2Vy
dGlmaWNhdGlvbiBBdXRob3JpdHkwggIiMA0GCSqGSIb3DQEBAQUAA4ICDwAwggIK
AoICAQCAEmUXNg7D2wiz0KxXDXbtzSfTTK1Qg2HiqiBNCS1kCdzOiZ/MPans9s/B
3PHTsdZ7NygRK0faOca8Ohm0X6a9fZ2jY0K2dvKpOyuR+OJv0OwWIJAJPuLodMkY
tJHUYmTbf6MG8YgYapAiPLz+E/CHFHv25B+O1ORRxhFnRghRy4YUVD+8M/5+bJz/
Fp0YvVGONaanZshyZ9shZrHUm3gDwFA66Mzw3LyeTP6vBZY1H1dat//O+T23LLb2
VN3I5xI6Ta5MirdcmrS3ID3KfyI0rn47aGYBROcBTkZTmzNg95S+UzeQc0PzMsNT
79uq/nROacdrjGCT3sTHDN/hMq7MkztReJVni+49Vv4M0GkPGw/zJSZrM233bkf6
c0Plfg6lZrEpfDKEY1WJxA3Bk1QwGROs0303p+tdOmw1XNtB1xLaqUkL39iAigmT
Yo61Zs8liM2EuLE/pDkP2QKe6xJMlXzzawWpXhaDzLhn4ugTncxbgtNMs+1b/97l
c6wjOy0AvzVVdAlJ2ElYGn+SNuZRkg7zJn0cTRe8yexDJtC/QV9AqURE9JnnV4ee
UB9XVKg+/XRjL7FQZQnmWEIuQxpMtPAlR1n6BB6T1CZGSlCBst6+eLf8ZxXhyVeE
Hg9j1uliutZfVS7qXMYoCAQlObgOK6nyTJccBz8NUvXt7y+CDwIDAQABo0IwQDAd
BgNVHQ4EFgQUU3m/WqorSs9UgOHYm8Cd8rIDZsswDgYDVR0PAQH/BAQDAgEGMA8G
A1UdEwEB/wQFMAMBAf8wDQYJKoZIhvcNAQEMBQADggIBAFzUfA3P9wF9QZllDHPF
Up/L+M+ZBn8b2kMVn54CVVeWFPFSPCeHlCjtHzoBN6J2/FNQwISbxmtOuowhT6KO
VWKR82kV2LyI48SqC/3vqOlLVSoGIG1VeCkZ7l8wXEskEVX/JJpuXior7gtNn3/3
ATiUFJVDBwn7YKnuHKsSjKCaXqeYalltiz8I+8jRRa8YFWSQEg9zKC7F4iRO/Fjs
8PRF/iKz6y+O0tlFYQXBl2+odnKPi4w2r78NBc5xjeambx9spnFixdjQg3IM8WcR
iQycE0xyNN+81XHfqnHd4blsjDwSXWXavVcStkNr/+XeTWYRUc+ZruwXtuhxkYze
Sf7dNXGiFSeUHM9h4ya7b6NnJSFd5t0dCy5oGzuCr+yDZ4XUmFF0sbmZgIn/f3gZ
XHlKYC6SQK5MNyosycdiyA5d9zZbyuAlJQG03RoHnHcAP9Dc1ew91Pq7P8yF1m9/
qS3fuQL39ZeatTXaw2ewh0qpKJ4jjv9cJ2vhsE/zB+4ALtRZh8tSQZXq9EfX7mRB
VXyNWQKV3WKdwrnuWih0hKWbt5DHDAff9Yk2dDLWKMGwsAvgnEzDHNb842m1R0aB
L6KCq9NjRHDEjf8tM7qtj3u1cIiuPhnPQCjY/MiQu12ZIvVS5ljFH4gxQ+6IHdfG
jjxDah2nGN59PRbxYvnKkKj9
-----END CERTIFICATE-----

Trying to add that gives a
`Das Zertifikat darf nicht länger als 2048 Zeichen sein!`

### To Reproduce Bug

Try to add the sectigo root certificate in the web ui.

### Expected Behavior

It accepts the certificate

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

64afbf8

tbnobody, 2023-01-17:
Servus Philipp,

RAM is limited. The certificate as to be hold in the RAM. So the first approach was to keep it as small was possible. I will add additional ~320 byte.

closed 2023-01-17 by tbnobody

pvogt09, 2023-08-08:
I have a self-signed root certificate that is around 3800 characters long, is there any possibility this might fit into RAM completely and work with OpenDTU?
When I paste it into the `textarea`, it just gets shortened to the maximum allowed length without any warning or further information.

stefan123t, 2023-08-08:
@pvogt09 is that cert self signed by your own CA or is this from a common vendor / supplier like eg Let's Encrypt. 
While I do see CAs moving from 2048 to 3072 and 4096bit RSA keylength we may have to adapt too.
Though we are short on memory and I would assume supporting the current default at Let's Encrypt should be okay/enough. Everything else would probably require a custom build I reckon.

pvogt09, 2023-08-12:
It is self signed by my own (selfsigned) CA and has a keylength of 8192bit. Ok, no problem.
Would you expect the remaining memory to be sufficient to run OpenDTU with a keylength of that size?

stefan123t, 2023-08-12:
Public sites (BSI recommendations) have switched from 2048 to 3072 or 4096 recently. I kind of doubt that this is necessary for a private Home Assistant or ioBroker which is running in your local network only, ie unreachable by / exposed to the Internet.
Please bear good measures

pvogt09, 2023-08-12:
It is probably not necessary, that's true. I run OpenDTU without any SSL at all right now and it is also not strictly necessary to change that, but it would be more of a "nice to have" and I was wondering why I couldn't configure it this way.
In case I compiled it on my own in a custom build, would you expect the memory to be sufficient to run smoothly?

stefan123t, 2023-08-13:
@pvogt09 short answer: No why would you.

Just because it might fit into memory, why would you require the low-level microcontroller (compared with a desktop pc / server) to do some additional rounds (4x the current max) to calculate the checksums for a protocol which you are not securing on the admin interface.

You wouldnt drive to the nearest gas station to fill your tank and drive four times around to always make sure the tank is empty before you refill it, while you theoretically and practically could ?

commit a4858b45 (2023-01-17): Fix #520: Allow larger certificates