tbnobody/OpenDTU #531: Reduktion Veröffentlichungsintervall minimal auf 1 Sekunde

HaraldNoack, 2023-01-23:
### What happened?

Bitte das Veröffentlichungsintervall auf 1 Sekunde einstellbar machen.

Die Norm schreibt ein maximales Ausregelintervall von 5 Sekunden bei "Nulleinspeisung" bzw. begrenzter Einspeisung vor.
Das Smartmeter hat eine gewisse Verzögerung bis die Daten beim MQTT-Broker sind und die Wechselrichter-OpenDTU-Kombination auch. Dann müssen die Daten verarbeitet werden und die neuen Grenzwerte an OpenDTU und Wechselrichter gesendet werden.
Bei einem 5-Sekunden Datenupdate ist somit die 5-Sekunden-Grenze nicht einhaltbar.

DANKE!!

### To Reproduce Bug

Veröffentlichungsintervall: 5....N Sekunden einstellbar 

### Expected Behavior

Veröffentlichungsintervall: 1....N Sekunden einstellbar 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

-

tbnobody, 2023-01-23:
Das Ganze hier ist so und so nicht Normgerecht. Hoymiles selbst sieht ein 15min Abfrageintervall vor. Ein 1-Sekunden Intervall ist alleine mit den Timeouts und CPU Auslastung nicht machbar. Hier kannst du nur versuchen ob es mit der Original DTU besser geht.

closed 2023-01-23 by tbnobody

HaraldNoack, 2023-01-23:
Danke für die Nachricht!!!

Und was wäre wenn die untere Grenze einfach fällt? Im schlimmsten fall ruckelt es oder stürzt ab und man kann sich selbst an das optimum ran tasten. 
z.B.: könnte man ja auch hohe Zahlen die sonnst nicht eingesetzt werden als kleine Zahlen interpretieren:
1...XXXXX in Sekunden und 100000....1000000000 in us und so könnte auch 1,9Sekunden as 1900000 (us) eingetragen werden.

So hat jeder die Freiheit zum Spielen.

Ich habe jetzt 10 Hoymiles im Einsatz und ein OpenDTU aber ich hätte auch kein Problem 10 OpenDTU für 10 Wechselrichter zu haben.

DANKE für dein SUPER Projekt!!!!!!

jonnycastaway, 2023-01-30:
Kleiner Tipp,
statt MQTT rufst du einfach die API ab.

http://<ip>/api/livedata/status

Das geht auch sekündlich. ;-)