tbnobody/OpenDTU #545: Power Value still shows up after Connection get lost

benedictjaeger, 2023-01-29:
### What happened?

If the inverter no longer has a connection to the OpenDTU, the last indication of power is still displayed until a new value comes in.

### To Reproduce Bug

Inverter is not available

### Expected Behavior

Set Power to 0W after 30 seconds or variabel time if no connection to the inverter is established

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

422d489

pijay01, 2023-01-30:
confirmed, that's happening to me as well. Would be great to get a variable after that the timeout get's reseted to 0. 30 Seconds could be too low.

muibusan, 2023-02-04:
Isn't it better to check the varibale "reachable" and as soon as it is "0" set the power to zero?#

And also compare https://github.com/tbnobody/OpenDTU/discussions/498

pijay01, 2023-02-04:
Your described way is a workaround, but I think this should be solved on the opendtu end. It should be fairly easy, check sunset, power and connectivity to the inverter and send a zero.

markusdd, 2023-03-28:
I can confirm this bug. This is especially problematic if data gets reported via MQTT because then ghost production is being recorded: 
[image]

tbnobody, 2023-09-07:
Duplicate of #498

closed 2023-09-07 by tbnobody