tbnobody/OpenDTU #549: [Request] Offset für yieltotal ?!

eeprom23, 2023-01-30:
### Describe the solution you'd like

Ist es möglich einen Offset für yieldtotal zu hinterlegen um eventuell gebrauchte Wechselrichter zu „Nullen“ oder sich den Jahresertrag besser darstellen zu lassen ?

jimknopf63, 2023-02-01:
Sollte möglich sein, Ahoy hat das mittlerweile implementiert. Da kann man einen Korrekturwert Yield Total setzen der dann vom tatsächlichen abgezogen wird

tbnobody, 2023-02-02:
Das YieldTotal wird aus Werten der einzelnen String Kanäle ausgerechnet. D.h. um einen Konstanten Datenstand zu haben müsste der Offset ja pro String-Eingang einstellbar sein. Ist möglich, aber nur damit das so bewusst ist.

eeprom23, 2023-02-03:
> Das YieldTotal wird aus Werten der einzelnen String Kanäle ausgerechnet. D.h. um einen Konstanten Datenstand zu haben müsste der Offset ja pro String-Eingang einstellbar sein. Ist möglich, aber nur damit das so bewusst ist.

Diese Funktion wäre toll. Bei ahoy ist es glaube gelesen zu haben auch so gelöst ! Vielen Dank für deine tolle Arbeit.

JoeGhost-22, 2023-02-05:
Ist mir heute auch aufgefallen, da ich eine ahoy laufen habe und sie dort vorhanden ist. Die Nullung wäre ein nice to have. Besser wäre wenn beide dtu's die gleichen Topics nutzen würden. Ansonsten top Arbeit.

tbnobody, 2023-02-14:
Implemented in 8cbae767970cc356026be3ac87279f3852d9b036

closed 2023-02-14 by tbnobody