tbnobody/OpenDTU #553: [Request] Adding communication via modbus

ArekKubacki, 2023-02-02:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

Is it possible to add Modbus TCP communication to the project?

Adminius, 2023-02-03:
what would you see on modbus?
As a client(read) or server(send)?

ArekKubacki, 2023-02-04:
Hello,
In the old nomenclature it was a slave.  
In the new one, it coincides with the server.
It would be nice if the addresses of the variables coincided with those of the original DTU, but it is not a must have.

Adminius, 2023-02-14:
Do you have any description for original DTU registers?

ArekKubacki, 2023-02-14:
I added pull request with modbus implementation :)

traeu, 2023-04-26:
Just started to learn about Hoymiles HM-600 and this project. It would be really cool if OpenDTU could communicate over modbus TCP using sunspec protocol. the original DTU-pro seems to support sunspec, but only over RS485*, I really can't understand why. I wasn't able to find any microinverter that speaks sunspec over TCP. A solution using OpenDTU would be great. 

*_"There are two types of modbus protocol on the DTU-Pro, in which one is Hoymiles Modbus
protocol and another is Sunspec Modbus protocol. Hoymiles Modbus protocol is developed
by Hoymiles and implemented on two different types of interface, which is Modbus RTU on
RS485 port and Modbus TCP on Ethernet port. You can find the detailed information in
chapter 4 and chapter 5. Sunspec Modbus protocol is developed by Sunspec Alliance and
Hoymiles also adopted this protocol on the RS485 port."_
https://www.shinetech-power.de/wp-content/uploads/2021/07/Technical-Note-Modbus-implementation-using-3Gen-DTU-Pro-V1.2.pdf

AloisKlingler, 2023-05-10:
Hello,

I think I can attach me to the question: are there plans to simulate an modbus-tcp energymeter on the device which can be simply added to e.g. a fronius inverter as additional "inverter-energymeter" ?

Thanks!

[image]

Sesshoumaru-sama, 2024-03-26:
Sounds interessting, would be nice if OpenDTU would implement SunSpec ... thats the last bit the HoymilesDTU does and if useful for integration in other vendors energy system (like Varta).

osulivanDE, 2024-04-21:
Sunspec Modbus would be perfect to be able to integrate Victron ESS directly via Victron GX device without scripting.

BjoernKellermann, 2025-04-23:
With SunSpec/ModBus over IP, OpenDTU could be fully used including Zero-Export with Victron Energy. See fronius-dbus which in fact is SunSpec.
https://github.com/victronenergy/dbus-fronius