tbnobody/OpenDTU #560: Some stats like Irradiation are not auto-discovered in Home Assistant

mathieucarbou, 2023-02-06:
### What happened?

When activating MQTT and HA discovery, some sensors are created. But some stats (like irradiation %) are missing.
We can by listening at the MQTT broker that they are sent to HA, but no sensor was automatically created.
This is annoying because for such specific cases we have to manually declare a sensor in our config.

I.e. for:

```
Message 774 reçu sur opendtu/112183226560/1/irradiation à 12:35 :
28.05
QoS: 0 - Retain: false
```

I have to manually set in the config:

```yaml
mqtt:
  sensor:
    - name: "Inverter 3 String 1 Irradiation"
      state_topic: "opendtu/112183226560/1/irradiation"
      unit_of_measurement: "%"
      state_class: measurement
      unique_id: "01e37eed-3045-4eee-a786-1249567fe700"
      device:
        identifiers: "112183226560"
```

so that I can expose irradiation 

### To Reproduce Bug

N/A

### Expected Behavior

Have the irradiation sensor automatically created like the other ones when mqtt is activated

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

422d489

tbnobody, 2023-02-06:
Did you enable the option to publish all individual panels?

mathieucarbou, 2023-02-06:
> Did you enable the option to publish all individual panels?

Oh boy! No... Actually I thought this option was not available since it was greyed out... Maybe red could be a better indication  for `off` ? ;-)

I am trying now...

mathieucarbou, 2023-02-06:
It works now ;-) Shame on me lol! 
Thanks!

closed 2023-02-06 by mathieucarbou