tbnobody/OpenDTU #571: Setting inverter limit needs about 25 - 90 seconds to be fully active

JsBergbau, 2023-02-07:
### What happened?

Setting the inverter limit either via MQTT (/cmd/limit_persistent_relative) or Webpage (HM-1500) it takes about 18 seconds until inverter reacts and begins adjusting the power. After about 25 seconds the inverter is using the new power limit fully, so it gradually adjusts the power over a few seconds. 
According to https://www.hoymiles.com/wp-content/uploads/2023/01/Technical-Note-Export-Management-Guide-of-Hoymiles-Microinverter-System_AU.pdf there is a Zero Export control with original DTU-Pro. So to support Zero Export the inverter must react much faster. I think that this faster mode can be enabled in the inverter otherwise "Zero Export" as Hoymiles advertises is no real zero export because load can change a lot in 25 seconds. 

With a Shelly 3-PM it can be verified that this delay is real and current output power display in OpenDTU is in sync with power Shelly 3-PM reports.

### To Reproduce Bug

Set the power limit and measure the time until new output power can be mesured.

### Expected Behavior

Output power is adjusted within 1-3 seconds.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

b5d099f

### Anything else?

Maybe this mode can't be enabled and it is a limit of the inverter, but than "Zero Export" as Hoymiles advertises is a marketing trick because it reacts much too slowly and there will be power exported and of course also imported, because increasing the inverter power also needs some time.

MQTT messages was sent at 12:50:15 to reduce power to 10 % (before 100 %)
Inverter began adjusting power at 12:51:40, so about 90 seconds later.
Until it was completly adjusted it was 12:53:21, so it took about 100 seconds to adjust power, together with the time until it began adjusting power about 3 minutes have passed. This is much too slow.

[image]

Simon566, 2023-02-19:
one question: does Ahoy DTU already have that feature , so it may be possible to take it over from there ? is that zero export function a secret ?

JsBergbau, 2023-02-19:
The question is if this zero export function is really faster working than setting the inverter limit. Perhaps it is just advertised as zero export but in fact inverterter takes also with official DTU Pro some time to adjust. 
I'm reluctant to buy a DTU Pro and check myself, because you also need a special energy meter to communicate with DTU Pro, see https://www.hoymiles.com/wp-content/uploads/2023/01/Technical-Note-Export-Management-Guide-of-Hoymiles-Microinverter-System_AU.pdf page 3. As far as I remember this is Modbus protocol. Of course you could also generate these packages to use Shelly 3EM to feed the current power values into DTU Pro, but also that requires a lot of time and I'm over 50 % sure that also with official DTU Pro inverter won't adapt its output power faster with Zero Export enabled in DTU Pro. So there is a high risk of investing a lot of time for nothing, because also official devices won't work better than OpenDTU.

Simon566, 2023-02-19:
today isnt sunny here , got issues to test it.

1. the new limit is taken over in 5s or so (HM300)
2. apparently the hoymiles is using a ramp up /down or so. If you put it to 100% (300W) but the weather doesnt allow it, it seems to ramp down percentwise somehow to the target. You wont see any power reaction until the percentage rampdown hits the real power produced and then it gets lower ( my impression )
3. when playing with percentage values BELOW the current power capacity , there is a quick reaction 
4. i do see something like 5W-10W reduction/increase in 3s, but i will retest when i get to a sunny day where it gets into its working range

-> it could be that the hoymiles firmware is ramping up/down slowly, but i have seen different reports in the internet about almost instant reactions ......

Simon566, 2023-02-20:
Hi,

i can confirm that the HM300 is ramping with about 5W roughly in 3s , starting from the previous limit

:-( thats slow

regards,
Simon

lawern, 2023-05-02:
In AhoyDtu project various ramp rates are described:

- [https://github.com/lumapu/ahoy/wiki/Protocol#wie-wird-das-down_pro-0x0e--down_dat-0x0a-verwendet]( https://github.com/lumapu/ahoy/wiki/Protocol#wie-wird-das-down_pro-0x0e--down_dat-0x0a-verwendet)

It would be great if we could set these parameters in openDtu to get a faster reaction.

Fribur, 2023-11-24:
Apparently the ramp speed to set a new limit can be configured to 15s (soft limit) or 2s (hard limit) as per [appendix  2](https://www.hoymiles.com/wp-content/uploads/2023/01/Technical-Note-Export-Management-Guide-of-Hoymiles-Microinverter-System_AU.pdf)

JsBergbau, 2023-12-13:
I understand this appendix different. Hardlimit disconnects the inverter and in the example the softlimit is 500 Watt, whereas the hardlimit is 1 kW

Page 12 https://www.hoymiles.com/wp-content/uploads/2023/01/Technical-Note-Export-Management-Guide-of-Hoymiles-Microinverter-System_AU.pdf
> In the hard export limit control mode, when the hard limit is exceeded, the export limit control function
> shall operate to disconnect the inverter or each inverter within the multiple inverter combination.

stefan123t, 2023-12-13:
@JsBergbau @Simon566 thanks for describing and measuring the ramp up / down rates with your installations and the Shelly 3EM.

The Ramp Up / Down Rates are defined per inverter in a so called Grid Profile. This can be set via the S-Miles Cloud and a DTU connected to your inverter.

We have an option in newer OpenDTU versions to extract the current Grid Profile in binary / Hex-Format from the UI. There are already about a dozen distinct Grid Profiles documented on the Wiki:

https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser

Thanks to @noone2k we already have a [python script](https://gist.github.com/noone2k/0b3a116a6f35286abef7199b62a0777a) which can be used to parse these Grid Profiles and make sense of the encoded settings within the binary code.

So it would be good to identify the values in your inverters for the Normal RUR_NM and Soft Start Ramp up Rates RUR_SS.

In fact there is a separate Guide from Hoymiles on the Grid Profiles which also goes into some depth of how the Ramp Up / Down Rates work and are applied.

https://www.hoymiles.com/wp-content/uploads/2022/11/Technical-Note-How-to-set-Hoymiles-3Gen-Grid-Profile-V1.1.pdf

So in general this issue / feature request is more of a long shot as we need to understand and implement setting the Grid Profile first, but the these parameters may be adjusted within the given regulations, laws and policies of your country.

closed 2024-03-21 by tbnobody