tbnobody/OpenDTU #573: Compiling ... resolving wt32_eth01 dependencies... fails

Maximilian4711, 2023-02-08:
### What happened?

Trying to compile 

[env:wt32_eth01]
; http://www.wireless-tag.com/portfolio/wt32-eth01/
board = wt32-eth01
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=4
    -DHOYMILES_PIN_MOSI=2
    -DHOYMILES_PIN_SCLK=12 ; 32
    -DHOYMILES_PIN_IRQ=36 ; 36i
    -DHOYMILES_PIN_CE=14
    -DHOYMILES_PIN_CS=15
    -DOPENDTU_ETHERNET

 aborts showing

Resolving wt32_eth01 dependencies...
Error: Could not find one of 'library.json, library.properties, module.json' manifest files in the package

Compiling same pin assignment with [env=generic] works fine on Win11 Visual Studio Code 

Version: 1.75.0 (user setup)
Commit: e2816fe719a4026ffa1ee0189dc89bdfdbafb164
Datum: 2023-02-01T15:23:45.584Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
Betriebssystem: Windows_NT x64 10.0.22621
Sandkasten: No

(but then image lacks Ethernet function) 

What´s going wrong with [env:wt32_eth01]?

Please help! 

### To Reproduce Bug

each compilation aborts

### Expected Behavior

success

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

?????

### Relevant log/trace output

```Shell
....
```

### Anything else?

How do I check the git-hash/version of OpenDTU I pulled?

tbnobody, 2023-02-08:
Works perfectly here... Based on the error I would assume that there is something wrong with your PlatformIO installation. 
You can also try to use the generic compilation and upload the device profile for the wt32_eth01 board... (pls see the documentation regarding device profiles.)

tbnobody, 2023-02-14:
I am going to close this issue as I cannot reproduce it and didn't get any response. Feel free to open it again.

closed 2023-02-14 by tbnobody