tbnobody/OpenDTU #587: [Request] Setting YieldTotal to a starting value

DarthWeber, 2023-02-14:
### Is your feature request related to a problem? Please describe.

after compiling a new version and uploading, opendtu starts again with yieldtotal of 0. Is it possible to rescue the old value ? My idea was to hardcode the beginning value with platformio but i  cannot find the right .cpp or .h file. Or is there any other way e.g. by the console ?

### Describe the solution you'd like

probybly just to know the correct file in the .cpp or .h file of the source

Solarteur, 2023-02-14:
In my estimation, "YieldTotal" is stored in the inverter and is "only" read out by OpenDTU. 
Was the inverter reachable for OpenDTU at the time of the "error"?
With my OpenDTU, the previous value of "YieldTotal" is displayed again even after a firmware update.

tbnobody, 2023-02-14:
> In my estimation, "YieldTotal" is stored in the inverter and is "only" read out by OpenDTU.

Thats absolutly right. It doesn't make sense to hardcode the value as it is invalid at each power cycle. It's also a bad idea to save it in the flash as it will wear it out over time. And as @Solarteur says, the value is stored in the inverter and read out as any other statistic value.

DarthWeber, 2023-02-15:
Yes, you all are right. Opendtu did report the correct value after illumination, so there is no problem anymore.

closed 2023-02-15 by DarthWeber