tbnobody/OpenDTU #598: Change Limit via Home Assistant has no effect

itn3rd77, 2023-02-18:
### What happened?

I wanted to change the Limit via Home Assisitant but it seems neither the sliders entities for relative limit adjustment nor the number entities for absolute limit adjustment have any functionality.

[image]

I changed the values and watched the topics via MQTT Explorer but nothing changes. Is this a bug or intentional feature and the sliders and number entities are only for representation?

### To Reproduce Bug

In the device page of the auto detected device in Home Assistant move the slide e.g. for "Limit NonPersistent Relative" and check the result via openDTU UI or MQTT Explorer. Nothinfg happend and after the next update the slider is back at 100%.

### Expected Behavior

Changing the values via automation or UI in Home Assistant should result in an actial change of the desired limit.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

9681ede

tbnobody, 2023-02-18:
Just doublechecked it here... works without problems:
[image]

[image]

First the command topic changes to 89 then after a little bit the status topic.

itn3rd77, 2023-02-18:
Thanks again for the quick reply. Double check ACL and permissions but did not find an issue. As you confirmed it is working I removed the cmd topic and and now it is working - crazy shit. Double checked after a restart of HA and MQTT broker. Still working. Don't know what was wrong with the cmd topic, but now it works without flaws.

Thanks and best regards!

closed 2023-02-18 by itn3rd77